f5bud/FX-4CR-V2 #1: S-Meter calibration

Starmike98, 2024-02-27:
My S-Meter is reading really low.  Is there a calibration with that.  I did the power calibration, voltmeter calibration, and watt meter calibration

f5bud, 2024-02-27:
Hi Starmike,
I'm working on it.
In a day or two there will be a new release.
I will let you know.
Can you give me a general opinion on this version.
Thank you, 73 Daniel

KX5SP, 2024-03-17:
Daniel - your V2 firmware is terrific. I notice on most out-of-band (MARS) freqs the power seems to be limited to 2 watts. Is this  intentional? There is one band just below 20 meters (around 13.6MHz) where it will put out a full 20 watts. Thanks.
73,
Steve KX5SP

f5bud, 2024-03-18:
Hi Steve,
Thank you for the compliments.
There is no function that reduces power outside the amateur bands. I think
it's due to the output filters.
You should draw the response curve to get an idea.
Be careful not to burn out the output transistors because apart from the
filters the VSWR must rise very quickly.

73, Daniel F5BUD

closed 2024-05-28 by f5bud