f5bud/FX-4CR-V2 #2: Feature request - TUNE function

KX5SP, 2024-03-18:
It would be wonderful to have a TUNE function that would put out a 5 watt carrier to tune antenna tuners. Button real estate is scarce but perhaps a long press AM/FM button?  Thanks.

Steve KX5SP

f5bud, 2024-03-18:
The design of the program does not make it easy to achieve this function
but I will try.  This is a function that seems logical to me.

Daniel

KX5SP, 2024-03-18:
Thanks Daniel. I'm liking your firmware more & more each day. You did a great job!

⁣
_____
Steve

f5bud, 2024-03-19:
Steve,
I added the tune function as you described it to me.
Can you test it and check that there is no interaction with the other keys.
I tested it but sometimes I don't see everything.
Daniel F5BUD

f5bud, 2024-03-19:
Sorry, there is a problem with the RIT as PTT function.
Take the corrected version.

Daniel, F5BUD

KX5SP, 2024-03-19:
Daniel - sorry, all I see on github is FX-4CR-V2-24-03-16.zip<https://github.com/f5bud/FX-4CR-V2/files/14634199/FX-4CR-V2-24-03-16.zip> which is the previous version?

f5bud, 2024-03-19:
The files are attached.  If they are not there, give me your email, it's
github that is blocking.

KX5SP, 2024-03-19:
Sorry no attachment. My email is:  ***@***.***   Thanks so much!

f5bud, 2024-03-19:
Gree ...
Replace # by . and * by @
daniel#nespoulous*gmail#com

closed 2024-05-28 by f5bud

sbridger, 2025-01-21:
In SSB mode, AMFM hold, makes radio appear to enter tune mode, power is set down to 5W, but there is no carrier - signal is still coming from microphone.

Also SWR values of 0.0 should be suppressed, the display should hold the last valid value.

f5bud, 2025-01-21:
I don't have this on my radios. When I press and hold AM.FM the radio
momentarily switches to CW and 5W

sbridger, 2025-01-22:
When in SSB mode, AND when holding PTT, AND AM/FM then PWR changes to 5W, but there is no carrier.

Expected:
Either: Let Tune mode output a carrier (my preference)
OR  Ignore AM/FM and do not enter tune mode at all.

---

Long Term:
I use tune out of band, and would like it to be minimal. I also need to tune an antenna standing up, or at the far end of the radio.

Tune Mode could use 
- Menu to set Tune power and continuous / pulsed.
- Pulsed: pulses the TX only long enough to read the SWR (e.g 5mS), then hold that value. Repeats every 500ms while in tune mode. 
- Walk away tune: Keeps pulsing. Encodes SWR as audio tone from speaker/bluetooth (bluetooth needs a better BT antenna to work 20m away)

f5bud, 2025-01-22:
Appuyez simplement sur AM.FM et maintenez-le enfoncé.  Le PTT n'est pas
utilisé dans cette fonction.