f5bud/FX-4CR-V2 #5: Feature Request: Forth

sbridger, 2024-09-06:
I suggest it might be useful to include Forth interpreter in the firmware. There is a [Forth for the StCube environment ](https://www.spyr.ch/twiki/bin/view/MecrispCube)& RTOS.

Why?
By putting Forth in, full (end-user) command of the radio becomes possible, without the main C firmware coder having to write all the possible special commands any user might want, in C. It also lets end users add a certain amount of functionality without having to make the C build environment work, and also frees the main developer from being asked to merge/alter the stable firmware to include user patches of unknown quality and stability.

As an example, CAT commands required that a special command interpreter was written to decode them. The full functionality of CAT could have been achieved in Forth, without any special command code being written at all.
 
Once a Forth interpreter is included, anything can be done, and for you the main firmware coder, it only requires linking your existing functions to the Forth if they are not, after that a requested new feature is up to the end user. You do not have to create and debug new functions that will only be used by a couple of people.

I imagine this would be mainly useful for enhancing remote control from the pc/phone, especially where people are making an app that needs "just one more command"
An example would be an app to sweep antenna SWR across the bands. 
Forth would allow an end user to write a command to do the sweeping, power changes, and return SWR data. Another might check all FT8 channels on all bands for signals as a propagation check.
Another example would be implementing the almost infinite permutations of automatically sending CQ that people like to have, or sending SOTA spots to the RBN.

(I am not currently an FX4CR owner, but opening the source has made it way more attractive to me. Schematics is the other important thing)

f5bud, 2024-09-08:
Hello,
thanks for the idea and the link.
I'll keep the information.
I don't promise to add this function immediately but I'll do some tests and
see how to adapt internal functions to forth.
73, Daniel F5BUD

sbridger, 2024-09-08:
Thanks Daniel,

Peter appears to be active with a calculator project he is working on at present (i.e. it's not dead), and he sounds approachable.
If it was me, I would talk with him first and discuss how to approach it.