f5bud/FX-4CR-V2 #9: Reduce RX power

sbridger, 2024-11-05:
RX draws 2950mW obviously via an SMPS as it is constant with supply voltage.
In my usage (7 day trips) it would be desirable to reduce this.

LCD probably draws 200mW-400mW (from random 2.8" lcd datasheets)

It would be interesting to get an idea where the power goes and how it could be reduced. 
Perhaps you can try some expts while doing development to see where the power goes?
Being able to select reduced performance in exchange for lower current when needed, would be acceptable.

- reduce cpu clock?
- eliminate spectrum scope ffts?
- lower sample rate?
- sleep LCD?

sbridger, 2024-12-28:
Having had a look inside, There are 3 usb IC's (hub, serial, codec), perhaps always powered off the radio's 5V

f5bud, 2024-12-28:
Good idea, I'll pass it on to Yu.
Thanks

sbridger, 2025-02-19:
I looked at the current when USB connected or BT on
Current stays at 0.23A@13V
Perhaps it is USB powered.

f5bud, 2025-02-20:
There is a problem with this set. The USB powers the bluetooth and the
bluetooth connects to the device even if the bluetooth is not active in the
menus.

sbridger, 2025-05-22:
Having used the radio for a while, the spectrum scope is seldom of much use to me, I would be quite happy to turn it off a lot of the time, for lower power drain.

Perhaps it should come on after band change or tuning for XXX seconds.
This assumes it uses significant power....

sbridger, 2025-06-12:
If sample rate was reduced, then the question is will the image rejection need to be adjusted.
Using my new filter, which has HPF at 7kHz, and is therefore quite phase active compared to stock,
at 12kHz IF, the image rejection only seemed to be 1dB worse @-54dB.
TLDR: TX calibration at 24kHz should hold at 12kHz

sbridger, 2025-06-15:
I found a transflective colour lcd, that might be a might improvement for sunlight and power

https://www.leadtekdisplay.com/24-inch-transflective-display-p-2849.html

Chip is allegedly ST7789T3.

I wonder how similar it is to the existing one sw wise?

f5bud, 2025-06-15:
Good news,
it's the same controller as the current display. We'll have to check with Yu if the dimensions and wiring are technically compatible. The price of the component also remains to be seen.

sbridger, 2025-06-15:
Size is different, I might see if it can squeeze in.
And just when I was planning to leave the case closed for a while...

sbridger, 2025-06-19:
Sunlight readable is the big (potential) advantage - the CPU seems to be the lions share of the power.

If it was fitted in, without moving the keys and viewport (only speaker has to move a bit), then 260 x 196 pixel would be visible.
Widening the viewport without moving the keys would give 280 x 222.
Changing the top plates machining without changing the pcbs or keymat might get you to 300 x 230

I might opt for a whole new top plate that overhangs the top of the radio allowing space for a battery pack. Then I would be able to get the whole display.

Anyway, interesting to know what they cost, and if they actually look any good in the sun.

sbridger, 2025-06-19:
Well a 2" one!
https://www.aliexpress.com/item/4001247598466.html

sbridger, 2025-07-02:
([Replying to this](https://github.com/f5bud/FX-4CR-V2/issues/33#issuecomment-3008683626) here as it's more relevant)
> The reference voltage is 2.5V. The internal layout of the SMT32H750 isn't straightforward. There's no indication that the internal reference can be output via the Vref pin.
> 
> I tried changing VREFINT but the final result didn't change.
> 
> SET_BIT(ADC12_COMMON->CCR, ADC_CCR_VREFEN); // 1: VREFINT channel enabled
> By changing VREFBUF, the DAC outputs were no longer outputting anything.
> 
> I took the opportunity to measure the consumption:
> 
> Mode	Consumption
> Power off	1mA
> Dfu mode	187mA
> Rx, speaker at 0	228mA
> Tx no signal	690mA

[This what I measured last year](https://reflector.sota.org.uk/t/fx-4cr/34090/18): 2950mW = 297mA @ 12.0V

So looks like a 20% improvement over Yus 1.7

As a matter of interest, 

Does it use the [cmsis dsp api](https://openaudio.blogspot.com/2016/09/benchmarking-fft-speed.html) for the dsp functions or C code?

I assume that the processor spends plenty of time asleep?

What is the clock speed?
What run mode is it in? [section3.5.3](https://www.st.com/resource/en/datasheet/stm32h750ib.pdf)

> Run mode (VOS0 to VOS3)
> – Scale 0: boosted performance (available only with LDO regulator)
> – Scale 1: high performance
> – Scale 2: medium performance and consumption
> – Scale 3: optimized performance and low-power consumption

Does it stay in one, or dynamically change? [section3.6](https://www.st.com/resource/en/datasheet/stm32h750ib.pdf)

sbridger, 2025-07-02:
I am thinking to connect the LCD backlight to the kbd backlights, so it can go off, and everyting associated with display can stop.
I wonder what the current drain is with no display spi, no display s/w updates, no ffts?

As an aside, I had trouble getting power drain to add up, unless t
The LCD+backlight is only ~500mW (80+20mA * 5V)
The max cpu current of 226mA @ 400MHz, all peripherals on, is only 1130mW. [Tables 127-129](https://www.st.com/resource/en/datasheet/stm32h750ib.pdf)
Not sure where the remaining 1300mW (264mA,5V) of 2950mW goes.

sbridger, 2025-07-09:
>The CPU is still running at 300 MHz.
[FX-4CR-V2-R-25-07-09.zip](https://github.com/user-attachments/files/21148910/FX-4CR-V2-R-25-07-09.zip)

14.310 USB 2.7kHz, low audio.
V 25-06-19: 226mA@12.0V (no change w/dsp-noise, CW/USB etc)
V 25-07-09: 221mA@12V DSP-Noise=0, 
V 25-07-09: 223mA DSP Noise=1

Doesn't look to me like cpu frequency has actually changed...

f5bud, 2025-07-10:
The CPU never sleeps. It takes a long time to enter and exit sleep. WiFi pauses the CPU, but it is continually interrupted.

sbridger, 2025-07-10:
> The CPU never sleeps. It takes a long time to enter and exit sleep. WiFi pauses the CPU, but it is continually interrupted.

I'm not seeing that the current is unchanged going from 400->300MHz, or going into sleep (WFI)
Sleep mode only seems to [take 3 cycles to exit](https://s-o-c.org/what-are-the-low-power-modes-in-arm-cortex-m4/)

Something is not right.

sbridger, 2025-07-14:
> The current display is a ZJY200K-IG01

Do you have the pinout or a datasheet for this? I'm trying to gate the backlight.

f5bud, 2025-07-14:
I don't know if the backlight control uses an AW9364.
To my knowledge, there's no connection to the CPU for PWM.
This could also generate unwanted signals in addition to the switching converter.

sbridger, 2025-07-15:
BTW, I have updated the [current calculations](https://github.com/f5bud/FX-4CR-V2/issues/9#issuecomment-3061601406) and they now add up to the actual current drain.

>I don't know if the backlight control uses an AW9364.

No. The LED's are directly across the 3.3V rail, with no series resistor. This probably explains why the radios current drifts around a bit.
They are drawing 83mA at the moment (it's cold), (169mA vs 216@12V)

I have added a PNP transistor off the keyboard BL signal to gate the LCD backlight LED_Anode, with a 47ohm so it is still dim (12mA) when off. This does mean, that I would like to set the keyboard BL timeout, as that seems to have gone.

>To my knowledge, there's no connection to the CPU for PWM.

No there is not.

>This could also generate unwanted signals in addition to the switching converter.
The 30MHz video data to the LCD is probably a far greater noise source. The LEDs are on the AMS1117 3.3V regulator by the BT module, not on one of the actual radio part supplies, so unlikely to interfere that way.
So potentially the KeyboardBL signal could be used for dimming.

f5bud, 2025-07-15:
Change "Back-Light" in menu to "Timed" and adjust "Menu-Delay"

sbridger, 2025-08-19:
Hi Daniel

- Is a tx bandswitch relay ON in RX now? [see](https://github.com/f5bud/FX-4CR-V2/issues/9#issuecomment-3063078554)
- Is the DSP done by ordinary C code or using [cmsis dsp api](https://openaudio.blogspot.com/2016/09/benchmarking-fft-speed.html)?

Profiling related questions:
- Can the FFT/spectrum scope be disabled for RX operation? (I mostly don't need spectrum scope when I need low power)
- Does screen update and lcd-drive use much cpu in practice? (I am turning the backlight off)

(btw, my experience is as other people have found - where the cpu cycles go is often/usually not where you guess, and you should use the profiler not intuition. FFT is a good example, where it often takes more time to process the result than it takes the highly optimised fft code to do the fft)

max-arnold, 2025-08-19:
Please do not disable the scope in RX mode completely without an ability to turn it on.

Btw, does the AM demodulation mode produces a good audio for you? On a couple of occasions for some AM broadcast stations I got better audio by switching to SSB mode...

f5bud, 2025-08-19:
Hi  Max,
No, I'm not going to disable the cascade!

The AM sound is significantly better when listened to with headphones.
In previous versions, there was a problem with the DC component obtained by demodulation.
The de-emphasis is EU-standard; perhaps this changes the tone a little.

73, Daniel

sbridger, 2025-08-20:
> * The screen refresh is performed in DMA mode (without CPU).
> The cascade calculation is performed using free CPU time.

It still burns up potential sleep cycles. I would not be surprised if drawing the waterfall and spectrum scope takes more processing than the fft.

Completely blanking the screen would allow:
- Stopping all the screen buffer redraw code, 
- stop DMA
- LCD controller stop or hold in reset
- fft & spectrum scope stops.
- lcd backlight off.

It would be interesting to know how much that can save, I am guessing at 60-80mA 12V saving, since [I see 46mA from the LCD BL](https://github.com/f5bud/FX-4CR-V2/issues/9#issuecomment-3071552403).

> Please do not disable the scope in RX mode completely without an ability to turn it on.

I was  thinking of disabling the waterfall/spectrum when the LCD is dimmed or blanked. Unless one wants to really save rx current, you won't blank the display.

f5bud, 2025-08-20:
I just launched the firmware without display and serial port management.
There is no FFT, no DMA transfer, the display controller is just hard reset.
The consumption is **193mA**.

The CPU shouldn't be dissipating much power because there's no heatsink on it.
The chip temperature stabilizes at 40°C while the shack temperature is 27.4°C.
Power is not wasted here !

f5bud, 2025-08-20:
Is it possible to attach a PWM to the KeyBL pin?

Yes, It's possible, I'll check. It's TIMER 4 CH4.
We'll have to see if it doesn't generate all kinds of noise.

sbridger, 2025-08-25:
It's an advantage really - key backlight brightness can reduce in concert with LCD brightness. 
In the case of night operation, both LCD and key BL are too bright, so it would be an advantage.

Not that that is my problem. My problem is that on long trips I want to leave the radio running on RX for an evening, or perhaps all day if it is a rain day. The lowest possible RX current is important for that - 200mA is less than 8 hours total RX in a week long trip for 1/2 of the 18650 battery energy.

sbridger, 2025-08-28:
> The small test program that varies the backlighting using PWM is also for experimenting with the screen backlighting. This signal should be used to switch the diodes on the rear panel.

I tested this controlling the LCD backlight and it worked with no visible artifacts.

> I was able to activate sleep mode when the CPU has nothing to do. (Very short time) This reduces the current under 12.0V by almost 20mA.

That's good. Perhaps worth revisiting stopping the fft/spectrum scope etc, Screenbuffer update, LCD drive, sleep the LCD controller IC.

f5bud, 2025-08-29:
By maintaining the reset of the ST7789 (LCD controller), consumption drops by 3mA.
The task that manages all the information on the screen consumes 3mA.
The FFT and spectrum part consumes 1mA of the 3mA.
Without FFT, there is no longer any Smeter or dB level.

At best, we can reduce consumption by 6 mA, which is negligible.

**The biggest saving is in the screen backlighting. A transistor must be added to be able to switch it.**

sbridger, 2025-08-31:
Now I have reduced the 5V rail to 4.553V , as nothing was gained by having 5V, and most of it is running off 3.3V regulators anyway.

Now, Ibright is 162mA, Idim=137mA. @12V

I have some backlight current as dim is done in hardware. Once backlight dim is done by PWM, I can turn the backlight off completely

>Without FFT, there is no longer any Smeter or dB level.

Once the backlight is turned completely off we do not need S meter or any other display functions. (stopping the display might also help with the patches of noise which are obvious in the 20-30MHz range)

> At best, we can reduce consumption by 6 mA, which is negligible.

It's still 5%. Reducing power is something that is done a few % at a time. It's approaching a 50% reduction of RX standby current relative to the original radio!

sbridger, 2025-08-31:
This is how I would like PWM backlight to work:
Backlight settings:

- BL Always ON / <Bright Time default 10sec>
- BL Bright %  (100% to 25%, default 100%)
- BL Dim % (100% to 5% default 25%)
- Display Off Time (0=never off.  0 to 16min in 30 sec steps)

All buttons, PTT, and frequency changes set display to bright. 
Only AF volume does not change the backlight brightness. You can change the volume without reactivating the display.
After Bright time the BL goes to dim %
After "DisplayOff time" the BL, LCD and all display function stops.
Menu time setting is unchanged. BL is always bright during menu time.

This has removed the "Backlight Always Off" option, as this could leave the LCD black and unable to change the setting. I'm not sure if anyone actually wants the keyboard leds to never be on, it seems pointless to me.

sbridger, 2025-09-19:
So [the new SOTA screen ](https://github.com/f5bud/FX-4CR-V2/issues/61#issuecomment-3303207136) is 138 on vs 135 screen blanked.
The waterfall screen is 138 vs SOTA screen at 138mA.
This tells me that the SOTA screen is still running the fft, as expected as you say this is where the smeter comes from.

These currents are my dim backlight currents, with reduced 4.5V VCC

f5bud, 2025-09-19:
To significantly reduce power consumption, all backlighting must be turned off.
In the “SOTA” version, when everything is turned off, the video controller is kept in reset mode and there is no more FFT or other calculations.
By adopting the diagram below, the screen LEDs are turned off. This avoids using another processor pin. 

[image]
(The diagram is simplified.)

sbridger, 2025-09-19:
I am using the KeyBL pin.
While RESET can be used to disable the lcd backlight, it cannot also be used to dim/adjust the backlight brightness, which is needed. 
Thus it is a better option to use the KeyBL signal which can both dim and disable the LCD BL with a single pin.
It also turns out to be fairly easy in practice on the pcb layout. There are issues about which tracks can actually be cut.
The other small issue using bipolars, is that the LCD backlight has no series resistor whatsoever. It directly connects to 3.3V. You lose a little brightness with even the 250mV transistor drop. A fet might be better. I used a pnp. It was OK for me.

f5bud, 2025-09-24:
Timer 4 is used for AF encoder pin 60, 61

sbridger, 2025-09-24:
I'd probably use a timer with two compare registers, each compare has an interrupt, one to set pin62, the other interrupt to clear pin 62.
But then I'm not an STM user...

max-arnold, 2025-09-24:
Maybe it is worth to inform Yu about these mods so he can integrate them into the next batches (or at least keep the pins unused).

sbridger, 2025-09-24:
Sounds like you'd be better to abandon the hardware and just use interrupts to control PD15. I think this is too hard a mod to recommend if there is any other choice. The LCD board mod is far less critical.

The timing is not critical, and it can be a low frequency 100Hz or higher. It worked OK in your test program.
(BTW, hope you removed the RX I/Q pull down R while you had it open. You might want to add the vox wire too)

>The problem is that the documentation is very confusing and wants us to use their HAL library.
Many programmers complain about this.

Yes the nRF bluetooth chips were like this - there is no datasheet with any actual register and hardware details. If the HAL routines don't work, you can't do it. We got 2 months into a project, too late to change cpu, and we had to stick an extra PIC in do the hardware bit since there was no documentation on the hardware of the shift registers.

sbridger, 2025-09-25:
An option is to just read the encoder pins like any other switch and resolve the encoder in sw. Then you could use Timer4 as pwm. 

My products settled on reading all switches and knob-encoders on a timer interrupt into a word of bit flags, and then calculating a series of derivatives. 
```
//timer interrupt routine.
This = read port pins into a word 
Changed=This XOR Previous      //set if input changed state
Set = Changed AND This        //valid until next read
Cleared = Changed AND NOT(This)
SetFlag = SetFlag OR Pressed    //persistent until cleared by the routine that actions it eg in main non-interrupt loop
Previous = This
```
By choosing the right sample rate this debounces dry contact switches. for this kind of encoder, somewhere 200 -1000Hz sample rate would do it.

---

>We should prepare a small electrical diagram with PC6 as the PWM for the lighting (keys and screen) and PD15 to control the key lighting.

Since that requires extra pins to the screen, he is likely to be reluctant.

sbridger, 2025-09-25:
OK tested. 
Based on current consumption, 20% setting is 22%
It flickers for one second out of 4 seconds.

I'll see how it looks outside over the weekend.

f5bud, 2025-09-25:
The advantage of using a timer to manage the encoder is that it works perfectly.
I can't manage the AF encoder by code and the TUNE encoder by timer. It complicates the code and it doesn't look good.

Furthermore, I can check the PC6 port to see if the modification has been made.
If it isn't present, the radio works as before.
Dimming works correctly with a wide range of light.

The next version includes the new code. The user manual has a section titled "Advanced Mode for Modified Radios" with minimal explanation. I'll add to it based on testing.

Version with timer as PWM:
[FX-4CR-V2-R-25-09-25.zip](https://github.com/user-attachments/files/22541460/FX-4CR-V2-R-25-09-25.zip)

sbridger, 2025-11-11:
Hi Daniel,

Havn't had any chance to get inside the FX4 for a while.
Is the PC6 pwm change in the latest FW?

f5bud, 2025-11-11:
No, the PWM modification for the backlight is not included in the published versions.