f5bud/FX-4CR-V2 #11: Feature Request: Report hardware revision on splash screen

sbridger, 2024-11-06:
Report hardware revision and serial number on splash screen
As the years goes by and fixes and mods are made, it becomes more of a pain when you can't easily tell what version of a radio you have or are about to buy.

sbridger, 2025-01-21:
In menu screen in V2.02

But can it detect the h/w version?

f5bud, 2025-01-21:
There is a bit that allows to detect the VR or R version.
Before any modification of the radio, I think that it is necessary to write
somewhere the version displayed when starting the radio with the official
firmware. It is a wise precaution

max-arnold, 2025-03-23:
Please add a hardware serial number somewhere. I need it to register my transceiver with the authorities to stay compliant with the ham operator rules in my country (most of commercially made radios have a s/n). Also someone on the FXradio group suggested that an s/n could be used by Yu to provide warranty for the rigs purchased directly from him.

I sent a patch to Yu in January, but it hasn't been integrated yet into the stock firmware. Here is the code you can use https://gist.github.com/max-arnold/ebcf09738d3bef876b3af1337cfb886e

And here is how it will look like:

[image]

f5bud, 2025-03-23:
There is a new update for serial numbers, look in release !
73 Daniel

max-arnold, 2025-03-23:
Thanks a lot! Will flash the firmware this week (I'm waiting for a power meter to arrive to calibrate the rig)

max-arnold, 2025-03-24:
I flashed the firmware and it looks like you used a different algorithm to calculate the s/n. Please document it here so I can communicate it to Yu. Otherwise when he eventually integrates my patch (that uses the official algorithm from the DFU), the displayed number will be different. This means it will be impossible to identify a device because its number will be firmware-dependent.

In other words, there should be some common ground.

f5bud, 2025-03-24:
I tried the "calculate_serial_number(void)" code, but it generated this number: 0038002AFFFF
I found it unsatisfactory and probably not unique.

The stm32h743/750 provides three Unique Device ID registers (96 bits) with the X and Y position of the chip, the wafer number, and the batch number.

The code used is as follows:
```cpp
char * getSerialNumber()
{
	static const char hex[] = "0123456789ABCDEF";
	static char buffer[16];

	uint8_t *id = (uint8_t *) UID_BASE;
	int16_t n = 0;

	buffer[n ++] = id[11];			// lot number ascii
	buffer[n ++] = id[10];
	buffer[n ++] = id[ 9];
	buffer[n ++] = id[ 8];
	buffer[n ++] = id[ 7];
	buffer[n ++] = id[ 6];
	buffer[n ++] = id[ 5];
	buffer[n ++] = '-';
	buffer[n ++] = hex[id[4] >> 4 & 0x0f];	// wafer hex
	buffer[n ++] = hex[id[4] >> 0 & 0x0f];
	buffer[n ++] = '-';
	buffer[n ++] = hex[id[2] >> 4 & 0x0f];	// y hex
	buffer[n ++] = hex[id[2] >> 0 & 0x0f];
	buffer[n ++] = hex[id[0] >> 4 & 0x0f];	// x hex
	buffer[n ++] = hex[id[0] >> 0 & 0x0f];

	buffer[n ++] = 0;

	return buffer;
}
```

max-arnold, 2025-03-24:
I agree that your version is better and directly maps to the unique device id. I sent it to Yu.

73, Max

closed 2025-08-31 by sbridger