f5bud/FX-4CR-V2 #13: Feature Request: Sunlight / high contrast screen

sbridger, 2024-12-28:
When using in clear sunny conditions, the large white text (e.g frequency) is readable, while the small non-white or low contrast is not.

A high contrast mode would be very useful.

Generally today's UI's have too much empty space between the informational text. The fonts are too small, and the font weight is too light. 
The use of fonts sourced from high resolution computer displays instead of low resolution optimised bold bitmap fonts makes them far less readable. 

Putting text inside buttons or boxes that do not add any information just makes the contrast lower, and reduces the text size.
Grey text on a grey background is totally unreadable in bright light: there is zero point in even having it on screen.

There are various tricks like having anything that is changing by knob or button, shown big while it is being changed. (The spectrum scope area could be used for this)

Given my use-case, if I had to choose, an ugly but readable screen would be my choice.

sbridger, 2025-09-18:
Reanimating this topic...

[image]

sbridger, 2025-09-19:
Hi Daniel,
Tested. I _really_ like the bigger clearer screen, even when I have glasses on. 
- I notice that no-signal rssi value seems to have dropped by 20dB from -134 to -154dB. 
- very short PTT action (one dit) does transmit, but doesn't wake screen from blanking in sota

---

Screen comes out of blank when Freq is turned as desired. 
I don't really want it to come out of blank when AF (volume) is changed, I can hear the volume change, I don't need to see it.

I suggest that "Sota Mode" is renamed to "Screen Layout: <Waterfall/Sunlight>" as the goal of it is clearer, and it's not really a SOTA mode as a lot of people will prefer a waterfall for SOTA.

I suggest that the status line should keep most of the same things: TX/RX, Voltage, AGC, Attenuator, BT (there is no point in an empty status line)

Showing temperature has no value to me as it is only cpu temperature (It could be hidden eg on the first menu screen with serial number). Showing Vol is of little interest also.

So if you move AGC and voltage back to status bar and drop temperature and voltage, you only need 8 values ie 4x2.
`NR     NT    POWER  BW`
`NB   TONE   WPM    RIT`

I suggest that the borders around values don't contribute information, by removing them you could increase the font size one step.
You could use a bold font as it has significantly more lit pixels - typically 30% more which makes bold 30% brighter than regular weight. Bigger font and bold might add up to 50% more lit pixels and brightness.
White will be slightly brighter than yellow, but perhaps not much brighter given eye sensitivity to blue is low. I would probably be happy with all white text, but hard to know until tried.

---
With a 4x2 grid, you could have a full width for CW decode - might be valuable when you do RX decode. 
Alternatively you could have a one line high spectrum scope with no waterfall, which is probably sufficient to see a signal appear in an otherwise empty band.
---

If you stick with 3x4 grid, Left/Right columns should be same sequence as the existing waterfall screen
Left Column:    ND,NR,NT.
Right Column: BW,RIT,PWR

sbridger, 2025-09-19:
While I have lots of suggestions, best to get something to try, and improve things when we see how it actually goes in use. (rather than spend too much time perfecting it.

Next suggestion is that you could compress the decimal point gaps in the frequency, and the put the VFOA/B/Memory# before the frequency as just A/B/NN

[image]

f5bud, 2025-09-20:
Here is another solution for SOTA mode (or screen-mode: 1 or 2).

I removed the temperature, which is no longer needed since the PA is on the outside of the case.

In practice, as soon as you touch a setting, the image with the parameters appears and after a second, the second image with the waterfall is displayed.
[image]

sbridger, 2025-09-20:
> I'm looking for a short word to replace "Sota Mode" which for me means outdoor activity.

Perhaps just "Sun Mode". alternatives: "Bold" or "Bright"

---

Playing in Paint, I noticed that while Yellow is 255,255,0 and Cyan is 0,255,255, it is still true that 
255,255,127 is still yellow, and 127,255,255 is still cyan. So you can get fairly close to white brightness, while still being coloured.
Ok, so yeah that should be obvious. I'll shut up now.

f5bud, 2025-09-20:
The screen is in RGB565 mode :)
with 5 bits for red/blue and 6 bits for green.

sbridger, 2025-09-20:
> Here is another solution for SOTA mode (or screen-mode: 1 or 2).

I am definitely preferring a direction of fewer and bigger. 

> I removed the temperature, which is no longer needed since the PA is on the outside of the case.

Ah, I wondered why it was there.

> In practice, as soon as you touch a setting, the image with the parameters appears and after a second, the second image with the waterfall is displayed.

I can't say if it would be better or a bit annoying, but definitely worth trying.

f5bud, 2025-09-21:
In this version, the "larger" display does not automatically show the settings banner. To display it, press and hold MENU.

I found this to be the most convenient option.

Next, I will change the “Back-Light” menu.
“Timed” will turn off the keyboard and screen with a timer.
“Always On” will reduce the keyboard by 25% with a timer.
“Always Off” will only turn off the keyboard.

This should be compatible with Simon's modifications.
Next, we will ask Yu to add hardware logic to turn off the screen with the controller's reset pin.

[FX-4CR-V2-R-25-09-21.zip](https://github.com/user-attachments/files/22450284/FX-4CR-V2-R-25-09-21.zip)

sbridger, 2025-09-21:
OK, trying it.

It has the ripple-down-the-screen refresh as it starts from sleep. i.e. it shows the screen, then a ripple rolls down the screen a couple of hundred milliseconds later.

I found the -20dB drop in rssi was caused by the notch filter being on. Which suggests that the rssi is not summing the energy of the whole bandwidth.

sbridger, 2025-09-21:
> “Always On” will reduce the keyboard by 25% with a timer.

... reduce the keyboard TO 25% , is what I am currently doing.  
It does need to be adjustable, as dim-but-usable depends on ambient brightness and your eyesight, it can't really be a fixed value.

> Next, we will ask Yu to add hardware logic to turn off the screen with the controller's reset pin.

LCD BL needs to be off the KeyBL pin. You can't use the RESET pin to control lcd BL brightness, and dimming the BL is the best way to save power for most users, because I don't expect that many people will choose to have the screen go totally black. For example doing SOTA or POTA, you would not really want the screen totally off. (I do, after a period of 1-4 minutes)