f5bud/FX-4CR-V2 #17: V2.02R Screen wraps / two button power on initialise bug

sbridger, 2025-01-21:
V2.02R Screen initialise bug.

When XX+PWR button combo is used to start the radio (e.g. CW+PWR or RIT+PWR) the screen Y/row origin is not getting set sometimes.

i.e. The whole screen is drawn, but the screen image is wrapped vertically i.e. the normal top of the screen starts part way down.

f5bud, 2025-01-21:
Yes it's a bug that has been around for a while. It can only happen with
PWR. From time to time I look for a solution.

sbridger, 2025-01-22:
Caught it today just with the PWR button.

Do these displays have a processor?
The old 44780 alpha LCD displays are in fact a 4 bit cpu inside. The initialise times are a giveaway that they aren't a pure hardware device. The clone 44780s were prone to having somewhat different and unstable timing and thus intermittent issues that required bigger timing margins than the datasheet said.

f5bud, 2025-01-22:
The screen controller is an ST7789. I programmed the SPI in DMA mode, which
transfers a 320x240 16bit memory page.
There are 4 tasks running in the radio. During the init phase, something
happens that disturbs the screen controller.

f5bud, 2025-01-23:
Problem solved 👍.
DMA1 is common to several functions: SPI, SAI, DAC. Each class initialized
the dma while it was already in use.
Added a flag to indicate that DMA is already initialized.

closed 2025-02-02 by sbridger