f5bud/FX-4CR-V2 #18: V2.02R When changing XX, PTT blocked until edit cancelled

sbridger, 2025-01-22:
When changing something eg TX Power, PTT is blocked while the edit line is on screen.

I expect PTT to work, and the edit to either be cleared immediately or just timeout normally.

Requiring that I put the PWR button a second time to complete the change is just an extra button push.

f5bud, 2025-01-22:
In PTT mode, you change the power directly by turning the AF knob.

sbridger, 2025-01-22:
This applies to all functions where the edit value is shown. (Filter, F Step, Sql etc etc)
The external PTT neither ends the edit nor begins TX.

Expected: When external or internal PTT is pressed TX begins, and current edit state is closed.

sbridger, 2025-02-03:
PTT clear menu works. Excellent!
 
But: Voice autosend now only sends once, then never initiates a repeat. The 4 lines of text remain. Pressing any of the 4 buttons initiates an immediate transmit.
I tried it for the first time yesterday, and it was working. (27Jan)
PTT now aborts autosend - which is what I would like to have happen. Yesterday I had someone call me back, just as autosend started up, and pressing buttons didn't abort it. Probably pressing any button or PTT during autosend should abort it.

closed 2025-08-31 by sbridger