f5bud/FX-4CR-V2 #20: Manual issues

sbridger, 2025-01-22:
(btw, happy to edit this directly if the file was somewhere) 

- Section 5.1  Phones & Speaker plugs DO NOT USE GND.  "AF" replace with "AF+" , GND replace with "AF-". Add the text "Phones & Ext Sp use push-pull amplifier. Do NOT connect sleeve to ground or excess current will flow and amplifier may be damaged."
- 
- Section 5.1 Add text  "Warning! PTT and KEY lines are connected to the CPU with no protection and no filtering. Be Careful: 12V here will destroy your radio. External filtering is needed with EFHW antennas to prevent noise and false PTT triggering."
- 
- Section 7 "Factory Reset" Should be described as "Factory Reset User Settings" and include the text "AB+PWR does not touch the calibration values"
- 
- Section 11.3 S-Meter Calibration mode. This is a little confusing. The software in cal mode, should be changed (from "dB") to display "dBm"  and the manual text to match: "-73dBm"
- 
- Section 3.11 RIT. Add text "RIT function must be set separately for every mode (LSB,USB etc)"
- 
- 3.12.1 - Volume adjustment. Currently says 
"There are two independent volumes levels:
1. When the sound is sent to the speaker.
2. When the sound is sent to the headphones."

I don't think that R hardware has the ability to know this. The ext. sp. and phones sockets are common.

- Section1 Specifications -> Receiving Sensitivity. Change to "-121dBm. 10m -116dBm; 6m -109dBm"

f5bud, 2025-01-22:
Thank you, I've incorporated your comments into both manuals. I will update
groups.io with FX-4CR-V1R-User-Manual
and continue incorporating in the zip for FX-4CR-V2R.
The manual is written in html and printed with chrome, which allows a
dynamic layout. I use a small web server to mask the url.

sbridger, 2025-02-09:
3.2.3 - Transmit power

Add: "While transmitting AF knob always adjusts power, without pressing PWR"

sbridger, 2025-07-13:
Notch Filter: Explain the notch centre frequency better, or just change the display to show the actual notch frequencies in Hz, so it is clearer what is happening.

sbridger, 2025-08-21:
I would suggest adding a file "changelog.txt" to the zip, which is just a plain text, cumulative list of all the changes you make.

It can be hard to find old versions before a change, or even know which changes have been made over time.