f5bud/FX-4CR-V2 #21: Feature Request: Hold SWR in SSB for stable readings

sbridger, 2025-01-22:
On SSB, the displayed SWR needs to be held when the level is too low for an accurate/stable measurement.
"SWR: 0.0" should not be displayed.

I would quite like the red TX lines - SWR/ALC and CW decode , to stay on for ~2 seconds after PTT/Key is released.

sbridger, 2025-05-09:
I notice SWR has negative values during ssb. I don't recall noticing it before.
(ver 30 Apr)

f5bud, 2025-05-14:
This is possible:

return (adcFwd + adcRev) / (adcFwd - adcRev);

if adcRev is greater than adcFwd, I'll add an abs()

sbridger, 2025-07-01:
> At sq2lip's request, I modified the SI5351 configuration to lower the reception to 200kHz.

I'll have to see where it goes.

> In the clock and PLL management, I modified the settings for low phase noise.
Simon, can you check the result? I don't know if my test is accurate, but the result is great.

What do you want me to check? The harmonic levels or the phase noise? What do you think has improved?

> The version of this message displays the three voltages: Fwd, Rev, and 12V.

Good, I will see if I can see the noise on it, but may need an actual noise value to make any progress.

sbridger, 2025-07-02:
RX gain relative to 14.1MHz 
200kHz -81dB
480kHz -57dB
1MHz -37dB
1.9MHz -5dB
2.2MHz  +1dB
3MHz -4dB
3.3MHz -1dB
3.6MHz +3dB

sbridger, 2025-07-02:
So I added caps to VRef, and got _more_ noise, and a zero offset count of 128.

Is it currently setup using the internal reference?

sbridger, 2025-07-02:
>VREFINT is enabled and set to 2.5V

I suspect that the TL431 and IntVRefBuffer are fighting (if they are both at almost exactly 2.5V), and adding C has made the instability more apparent. 
If you could make an ext Vref or switchable version I could compare them without having to unsolder the TL431 & R, which is my next move.
It is possible that setting vrefbuf to 2.048V would stop the TL431 from acting (as it is now <2,5V), and the VREFBUF might be able to overdrive the 300ohm pullup to 3.3V (2.6mA)

> I think both have been improved.

I looked at 14.1MHz -24kHz reciprocal mixing pn and it is unchanged (but it comes from the r&s).
Is 14.17 a known problem spot? 
Do you have plots of before, at the same freqs?
Are there known exact frequencies where the 5351 will have poor pn, or spurs?

The PN I can easily measure is by rx reciprocal mixing at +-24kHz. This level is really low at 14.100: 40dB better than  my R&S, (or about -100dBc in 128kHz RBW)
I must admit to being astonished by that.

Harmonic levels should be driven by levels in the p.a. chain. (or spec an input distortion). Remember the modulators are driven by square waves at the carrier freq, so the ssb starts out with high odd harmonic levels before it is amplified at all. (Harmonic levels might also possibly be affected by how tight the case screws are, due to the anodising and poor contact)

I think that what you see around your carrier and above DC is likely your spec an performance, not the radio.

sbridger, 2025-07-02:
Possible the TL431 pullup R overrides it.
Is 25-07-02 a 2.048V version?

I'll try lifting the TL431 off and see what happens

sbridger, 2025-07-02:
Previous test versions have ~200mV fluctuating on VREF
This version 2-Test-25-07-02 has 2.5001V on VRef
With 0.665V on FWD diode, I see 8623-8639
0 at 0 V
12.45V sense 32402 - 32483

Currently has 2x 100nF caps on vref.

sbridger, 2025-07-09:
I'm sure I have posted [this link, where he claims to have implemented many phase noise reducing tricks.](https://github.com/pavelmc/Si5351mcu) previously. But here it is (again), in case I didn't, and if anything in it is of interest

f5bud, 2025-07-10:
Yes, I use the same techniques.