f5bud/FX-4CR-V2 #26: Development/debugging process

max-arnold, 2025-02-02:
Hello!

I'm totally new to the STM32 embedded development process, but have the FX-4CR transceiver (the latest R revision) and want to try adding some small improvements to the original firmware. E.g. like displaying a device serial number on the splash screen and maybe add the 500Hz tuning step.

Could you please describe your development process?  Do you just use the DFU mode, or there is a way to have the STM32 debugging features enabled?

sbridger, 2025-02-02:
FYI, V2.02R 31Jan has 500Hz, 5kHz etc

f5bud, 2025-02-03:
Hi Max,

embedded development is complex access.
Either you use HAL libraries but it is not very efficient, or you write
everything. It is "bar-metal".

Use the stMicroElectronics IDE and try to make a test program that flashes
the keyboard.
To find the use of Gpio look in the source of V1.
The radio goes into dfu mode with F + PWR. There is no debugger.
To trace your program you have to configure Uart1 and do printf.

It's a bit of the stone age 😂

73, Daniel

max-arnold, 2025-02-03:
Thanks for your reply!  I hoped that it is possible to use ST-Link for debugging purposes...