f5bud/FX-4CR-V2 #27: Feature Request: Support S+N/N measurement

sbridger, 2025-02-03:
Try to measure S+N/N, without a sinad meter, (i.e by using an audio voltmeter) the AGC is active from very low levels and stops a valid measurement.

Options are:
- Use the ServiceMode->RSSI-CAL  displayed dBm value by having much longer averaging of the signal value and 1 decimal resolution in the displayed dBm value, so that it is a stable value. The displayed dBm value would be used instead of an external meter.
- Have an AGC OFF mode, perhaps in the service menu, so an external meter can be used

priority: not very

f5bud, 2025-02-08:
Hi Simon,
I made major changes to the ALC and AGC. Can you do a test?
I am reviewing the AM modulation. It is mainly the emission that is not
compliant. I will have to do the FM as well.
There is an OM who is doing research for a high-performance noise reducer.

73 Daniel

sbridger, 2025-02-08:
No audible audio noise at all. at power on. AB+PWR no effect. CW/USB/FM/AM no effect.
Signal level on display -200dB
Att button no change - displayed number changes from 8dB to 30dB
 
14.31 -73dBm RF, If ATT=30dB, level=S2
AGC is incredibly slow. Takes about 1 min for gain to be restored to S8
 
On ATT=8dB, no audio, no noise.
signal level is very slowly rising from -200dB at 1dB / 10 seconds
 
After a couple of minutes, switch back to ATT=30dB
It begins it's slow rise again. (rises much faster than at ATT=8dB
 
Switch back to ATT=8dB. Displayed level has gone from -140dB back to -180dB, while I have been playing with ATT=30dB
 
So:
1) Something wrong - not usable
2) Some AGC variable might be uninitialised at power on
3) ATT=8 and ATT=30dB agc accumulator seem to keep running when not selected

f5bud, 2025-02-09:
ok, I have two "R" radios, one works fine and the other has the fault you
have!!!
I'll see what causes this problem.
73

f5bud, 2025-02-09:
Among the studies, a function under development remained active, it was the
DNR.
Do a factory reset by default
Github is updated.

sbridger, 2025-02-09:
Audio is back.
When RF is -73dBm, pressing ATT results in a large click. (larger when entering 30dB than when ->8dB)
(large clicks are an issue with earphones, not so much with the internal speaker)
I am thinking that the signal level should compensate for the ATT value ie the indicated signal level remains constant.

ALC/MicAGC seems much improved on SSB.

sbridger, 2025-02-09:
I have a sense that AGC should have a threshold (e.g. -110dBm) before it acts. In conventional radios, agc acting too early tends to make the SNR worse. I'm not sure if that is happening here, it's just that the 10dB SNR point on the sig gen sounds noisy. I noticed this the other day, when I had the FX4 next to the KX2. The same weak CW signal seemed less noisy on KX2.
But it could equally just be my imagination.

f5bud, 2025-02-09:
I set the AGC threshold before saturation. I can't set it higher.
Should I provide an AGC delay setting in the setup for small, middle, fast?

sbridger, 2025-02-10:
>I set the AGC threshold before saturation. I can't set it higher.

I mean at the other (low) signal end. The threshold or control level for the AGC. Rather than the one at the clipping end.
We may be talking about different parts. The AGC effect I'm talking about might be happening in the post-detection / post-IF audio part.

The AGC seems to act from very low levels - probably within the noise floor. At -125dBm, the AGC seems to be reducing the audio output.

>Should I provide an AGC delay setting in the setup for small, middle, fast?
No what I was expecting was that the audio output level would increase linearly for the first 15dB before the AGC started cutting it back.

I am not sure that this is actually beneficial without trying it

sbridger, 2025-02-10:
What I had in mind is that gain remains at max until (say) -110dBm.

sbridger, 2025-02-13:
It would help to have AGC-OFF mode to be able to measure the effect of AGC on SNR, rather than just talking about it.

Options for the UI might be:

-Add an AGC-Off/On menu item to the service mode menu, that turns off AGC, only for service mode, or only when in this menu item.
-Add an AGC-Include-OFF menu item to the service mode menu, that adds OFF to the AGC sequence. i.e. the normal AGC selections becomes "OFF,FAST,MID,SLOW"
-Just add OFF to the AGC sequence always i.e. the normal AGC selections is always "OFF,FAST,MID,SLOW"

sbridger, 2025-02-13:
I managed to get [Spectrum Lab](https://www.qsl.net/dl4yhf/spectra1.html) running with the FX-4 and measuring the 10dB SINAD.

USB, 2.4kHz BW, USB Aux OutLevel=20, Sample rate=22050.

V2.1(9Feb)  -113dBm @3.620MHz (avoiding spurs)
V1.7 -124dBm @3.67MHz (using Comtekk SINAD program via headphone out and laptop mic in jack)

So sensitivity seems to be 10dB less

---

SpectrumLab must take a prize for arcane UI.
However it is free, can do 2tone, single tone, and SINAD, and RTS PTT control.
QuickSettings->Radio Tests->SINAD

sbridger, 2025-02-14:
I have [HF Radio Systems & Circuits](https://digital-library.theiet.org/doi/book/10.1049/SBEW011E). It's everything about SSB by Collins radio staff.

[AGC excerpt](https://www.ab4oj.com/icom/agc.pdf)

You can probably access a [pdf online](https://digital-library.theiet.org/action/ssostart?redirectUri=%2Fdoi%2F10.1049%2FSBEW011E_ch8) if you have access to a university library that subscribes to IETF...

sbridger, 2025-02-18:
> I noted the gains of the different stages: At -120dbm the gain is at the maximum. From -80dbm the sound gradually decreases (by ear). 
The AdcGain is an amp at the input of the ADC. It is an AD-605. 

I see that the CM8978 has 
-12:+35dB in the PGA, then 0/+20dB boost gain at the mixer.
This is a total of 67dB gain adjustment before the ADC.

>The DacGain is a multiplier by calculation before the DAC. 

Is this in the STM software, or it it the block in the CM8978 labelled "Digital Gain" ?(fig24)

f5bud, 2025-02-18:
I use the wm8978 at a minimum.
No filter at this level or gain variation during use.
The reason is that it is overclocked and the internal dsp part may not work
as expected.
There is a wiring problem that creates a DC. The wm8978 has a function to
counter this but it does not work, so I did not go further.

To establish the action levels of the AGCs, I blocked the gain of AD605 at
max, the digital gain DAC at minimum and I read the digital ADC values by
gradually increasing the signal at the input with an HP attenuator and a
tinySA as generator.
This gave me the saturation threshold of the input stages.
Then I did the same for the DAC, minimum gain on the AD605 and maximum
digital DAC gain (500)

f5bud, 2025-02-21:
In the same conditions and settings, I made a capture of a small gnuradio script of a CW signal at -127dbm.
(TinySA -7dbm, power attenuator -30dbm, HP attenuator -90dbm)
Version "V" the level is -36dbm, the floor is at -45dbm.
Version "R" the level is -32dbm, the floor is at -46dbm.

[image]