f5bud/FX-4CR-V2 #29: LOCK indication

sbridger, 2025-02-09:
(V2.1 Feb 9. Not sure if this is a recent change)
When LOCK mode is engaged the top bar changes from blue to dark red.
This is not visible in sunlight, and there is nothing that says LOCK, and the F button has no lock symbol, so a user could easily be left with a locked radio, and no idea why or how to unlock it.

Expected: The word LOCK on screen, or the padlock symbol. (I prefer the word LOCK)
Perhaps the currently unused VFO frequency should be replaced with the work LOCK - the user doesn't need to know what it would be because he cannot change it during LOCK anyway.

sbridger, 2025-02-19:
(V 17 Feb)

Exactly as I want. 
Fixed.

I notice (and I assume this is not a change) that neither RIT-PTT or Ext-PTT works in locked mode.

My expectation is that (like my FT817,891) you can still TX with a locked radio - just not change frequency, modes etc.

f5bud, 2025-02-25:
Now works the same as "the big ones"

OH2GT, 2025-02-26:
Great progress with the latest FW, Merci Beaucoup.

RIT button defined as PTT works fine in normal operations but not in Service Mode. Is this correct or maybe a minor bug?

f5bud, 2025-02-26:
Service mode is a bit special

I preferred to simplify the code rather than manage the RIT key.

The microphone PTT works.

OH2GT, 2025-02-26:
No probs and agreed. Let's keep it like it is.

closed 2025-06-12 by sbridger