f5bud/FX-4CR-V2 #31: ALC for FT8

sbridger, 2025-02-09:
(V2.10 Feb 9)
Using FT8TW from my phone (bluetooth).
AUX IN LEVEL=100
When audio level is high (85% on phone app), output power is low (2W). When audio level is low (18%) TX power is nominal 
(5W). Not sure if this is ALC as such, or clipping.

Displayed ALC value is 100% in both cases. 
Below 18% audio, the ALC value and power output both drop, as expected.

If I change AUX in level to 50, and phone out to 100%, the power out remains at 2W, and ALC remains at 100%

The issue is:

- High audio level sees RF output be low
- There is no indication that audio level is too high. You don't know to adjust the output level or the aux level.

Expected: Needs an indication that audio level is too high and must be reduced, at least for DIG modes.

BTW, Output power is no longer pumping like previous versions (stepping from 5W to 0.5W at 200ms). This version should make FT8 work much more reliably for people.

f5bud, 2025-02-09:
don't forget the bluetooth adapter that is in the radio, it also has its
alc.
Lower the level on the phone and adjust on the radio

sbridger, 2025-02-09:
>don't forget the bluetooth adapter that is in the radio, it also has its
alc.

I don't think that is the cause because the fx4 ALC indication stays at 100%. 

OK, so (3.5MHz,5W,10.2V, auxin=50) I looked at the rf envelope with an oscilloscope. 
As the phone audio out increases, ripple appears with a period of roughly 2ms.
95%ALC@ 10% "Signal output strength" in app on phone
100% ALC TX=37dBm @ 12% 
No/small ripple, 37dBm @ 44%
16dBm w/ripple@45%
20dBm w/ripple@46%
24dBm w/ripple @48%
27dBm w/ripple @ 52%
32dBm w ripple @67%

The ripple/low level can be seen rising up i.e. it is associated with some sort of time constant eg ALC action.

The question is whether it is something in the radio audio path or it's coming from the BT.
I don't think it is from the BT, as if it was the BT reducing it's output, the TX ALC would compensate. i.e. we might see the envelope ripple, but not the low output level.

Given that this shows an abrupt drop and ripple at an exact point, I think it is probably an ALC bug or instability. 
Probably there in SSB too if I put an audio sig gen in.

>Lower the level on the phone and adjust on the radio

The problem is that the ALC value displayed doesn't help you set ALC, because it is either 100% or it is not. There is no indication that the level is too high. It's not clear that you need to lower it. 
On other rigs there is a mic level meter and an ALC meter. I am reluctant to suggest that, because it's a real pain to switch meter functions. Dual meter bars i.e. show both mic and alc levels at the same time might work.

Perhaps the ALC value should go to numbers greater than 100% when level is too high. eg it stays at 100%

sbridger, 2025-02-09:
Using an audio coupler from a 1kHz sig gen, I don't see this sudden collapse of the level as audio increases.
As it overloads the peak envelope level stays constant and flat, and 1kHz AM waveform starts to appear.

sbridger, 2025-02-09:
OK Using USB cable, WSJTX 2.61, no cat, RTS TX, TUNE button.

As I adjust the level up it is the same as with BT. Hits 100% ALC, 5W, stays flat at 5W, abruptly drops to 20dBm at ~70% indicated on the slider,
As the drive level is incresed the power out increases.
There is an exact audio level where TX power drops to a wildly varying zero to 10dBm

f5bud, 2025-02-10:
I use a HF detector (1 schottky diode and a filter capacitor)
With gnuradio I made a two-tone sinus generator with variable amplitude.

The radio set to 10W, I get 31V on my detector.

On the USB inputs I send a BF signal strong enough for 100% ALC.
I get 10W on the radio screen and 31V on my detector.

On the USB inputs I send a BF signal with two frequencies (500 and 1400hz)
strong enough for 100% ALC.
I get 5W on the radio screen and 31V on my detector.

It seems that the radio power detector is no longer there.
When listening on another radio the signal is correct without distortion.

What is the problem?

sbridger, 2025-02-10:
5W is the average power, not the PEP (peak envelope power). 
The meter is reading average. The bar graph should be showing PEP
The ALC should be keeping the instantaneous peak power constant, not the average value.

ie V500+V1400=31V  V500=V1400=15.5V,  PavSingle=31^2, Pav=(15^2) * 2

---

To see my FT8 ALC problem make a single FT8 tone 1500Hz, via USB. Look at the RF envelope with your scope.
Now slowly increase AF from the point where it hits ALC=100%
ALC remains at 100%, and RF is constant with no envelope ripple, until you increase to about 4x amplitude (on my phone) when it abruptly drops very low and gets ripple.

f5bud, 2025-02-10:
With a single tone I can put a very strong signal and I still have 10W.
The calculation of the ALC and necessarily a peak detector. std::fabs() is
the equivalent of a diode

sbridger, 2025-02-11:
Hi Daniel. 
To make ALC issue more reproducible here is exactly what I am doing. 
I use a [small Windows free tone gen program TONE.EXE from Tolvan.com](https://tolvan.com/index.php?page=/tone/tone.php) to generate the USB signal. I am looking at the RF output envelope using a 20MHz analog scope in normal trigger mode. (timebase 1ms/div - 50ms/div)

So using TONE, 
F=1500Hz single tone (USB)
10.2V, PWR=5W (10V for less heat)
10.1MHz, Dig-U, RIT=PTT, Aux-In-Level=50  TX activated by pressing RIT/PTT

TLDR: 
- At -10dB USB level the output is normal.
- At -9.8dB the level drops to near zero
- This usb level does not change with AuxLevelIn
- At all usb levels, the output level overshoots to 14W with a damped oscillation of ~1ms period taking 3-5ms to stabilise at 5W
- changing V+ to 13.4V increases the overshoot to 3/1.4 = 23W peak.
- The overshoot issues previously seen using voice on much older versions of F/W probably still exist, the period of the instability and the damping time have reduced. 

Aux-In-Level=50

Level=-23dB, ALC=92%, Po=35.9dBm  RF envelope has no ripple.
Level=-22dB, ALC=100, Po=36.7dBm
Level=-10dB, ALC=100, Po=36.7dBm

(Level=-10dB) At PTT on, RFout rises to 2.2 on scope within 500uS, this is a damped oscillation with ~<1ms period for several cycles as it settles to 1.3 (5W,36.7 dBm) in about 5ms. The peak is (2.2/1.3)^2*5W=14W

Level=-9.9dB Po=36.7, but dropping out
Level=-9.8dB, ALC=100, Po=3dBm noise, bursts of 15dBm RF about 700uS wide. When PTT is pressed Power spikes as above at -10dB, then flat at Po=5W, then it drops off after 300ms down to near zero.

Level=-9.7dB, ALC=100, Po=12dBm When PTT is pressed Po=5W, flat, then it drops off after 200ms.
Level=-9dB, ALC=100, Po=24dBm
Level=-8dB, ALC=100, Po=26.7dBm Has ripple with period 0.7mS and amplitude of 1/5th (0.2)
Level=-0dB, ALC=100, Po=32dBm has ripple, and a slow level ripple at ~1sec period as well.

Now change Aux-In-Level to 100

Level=-29dB, ALC=92%, Po=36dBm  RF envelope has no ripple.
Level=-28dB, ALC=100, Po=36.7dBm
Level=-10dB, ALC=100, Po=36.7dBm
Level=-9.8dB, ALC=100, Po=3dBm

Now change Aux-In-Level to 200
Level=-10dB, ALC=100, Po=36.7dBm
Level=-9.8dB, ALC=100, Po=3dBm

sbridger, 2025-02-11:
>I use a HF detector (1 schottky diode and a filter capacitor)

Do you have a scope to look at the RF envelope?

If you are using a scope/soundcard/meter and the HF detector, you have to make sure that the time constant of the detector Rs and Cs (1/(2&pi;RC)) is fast enough for what you want to see. My default diode detector only has about 20Hz freq response.

f5bud, 2025-02-12:
> make sure that the time constant of the detector Rs and Cs (1/(2πRC)) is

Yes, I think the time constant is sufficient. I see the envelope of the
voise. In CW I have a horizontal line. In two-tone I also have a horizontal
line

sbridger, 2025-02-13:
>In two-tone I also have a horizontal
line
In 2 tone you should be seeing an AM envelope of the beat frequency between the two tones. (or the half wave rectified version)
https://www.youtube.com/watch?v=3DxBg6h4Fc8

If you have a diode, 100nF capacitor and oscilloscope (1Mohm), F<sub>lowpass</sub>=2Hz

sbridger, 2025-02-17:
>If I increase the signal above 1 in the audio chain between the computer
and the ADC input which saturates or limits and which gives an inverse
power of the signal.

Do you mean that an amplifier is inverting between the BT/CM108 and the ADC? (a common effect in opamps when one input exceeds common mode range)
How did you discover this? Can you see the ADC values?

>The more we exceed this threshold of 1 and the more the power decreases. 

This isn't quite what happens with a constant signal (which FT8 is). At a level just above -10dB (I assume this is relative to full output of the CM108, 0x7FFF-0xFFFF) the output is cancelled down to zero immediately and almost completely. 
FT8 won't work.

---

Looking at the schematic and datasheets.
CM108 has full scale of 1Vrms, and the AUXR input of the WM8978 has full scale of 1Vrms. 
We should not have overloading, if the gains/amplifiers are set right.
It appears to go wrong at -10dB at the CM108, suggesting that the gain WM8978 aux channel gain is set higher than 0dB
The AuxR (CM108) and R2(BT) inputs don't go through the PGA, but they do seem to have a +6dB amplifier
Looking at the block diagram pg14, R2 can be routed either through the PGA or to directly to the mixer gain block.
If there is an overloading amplifier, then perhaps it is being routed via the PGA when it should go direct to mixer.
The direct mixer input also appears to have a +6dB gain amp (R2_BOOST_VOL). Perhaps this is not 0dB or perhaps it should be -xdB until the overload stops?
(interesting that it has mic alc in the block diagram pg14)

It seems that the BT/USB inputs have too much gain, and it should be possible to configure to not have the catastrophic overload at -10dB. i.e 0dB to the CM108 = full scale of the ADC.
---

>adjust the signal power to wait for 100% of the ALC and not go beyond.
The issue here is that there should be a clear indication of overload. Normal radio meters show the amount of ALC overdrive, and have a mic level meter as well.
Having the ALC value top out at 100% blinds you to this.

Actions:

- Make the ALC indicator show levels above 100% rather than clipping
- If the BT/USB gains cannot be corrected to prevent overload, then have an overload indicator. The current -10dB point is catastrophic to the FT8 signal and a silent fault.

f5bud, 2025-02-17:
I will check tomorrow morning. Currently AUXL2BOOSTVOL is at 0db.

sbridger, 2025-02-18:
The very incomplete schematics I have don't show coupling caps.
It is possible that there is a DC offset between the CM108 and BT outputs, and the WM8978 input. i.e. V<sub>mid</sub> of the MC108 is different to CM8978.

f5bud, 2025-02-18:
> and the WM8978 input. i.e. Vmid of the MC108 is different to CM8978.

Yes it is in my previous answer

sbridger, 2025-02-18:
I was thinking of the USB signal from the CM108, not the AD605 signal.

Here is mine. 

[image]

Looks like the mode pin (18) has been changed with the added resistor. 
Also looks like the input pins for the AD605 are different too. (L2,R2)
And the pulldown R value might be different - 100k on mine. 
Datasheet says Rin is 94k so V will be almost half way to gnd. 
WTF?

f5bud, 2025-02-18:
the resistor on pin 18 changes the operating mode of the wm8978 to i2c or
spi. This is a logic level. If there is sound it is good!

f5bud, 2025-02-18:
I don't have a schematic of the aux or mic inputs.
On the latest revision "R" the usb and bluetooth inputs are mixed together.
There is a DC component on all inputs.
The most important is the one on the IQ signals.
It limits the amplitude of the reception signals.
I made a schematic to understand the wiring on the WM8978.
This is what allowed me to program the registers more easily.

But I don't understand where the problem is with the usb or bluetooth and even microphone inputs.
You have to adjust the audio level of the computer or phone so that the ALC indicates 100% and not exceed it.
From there, with a one-tone signal the power indicator is identical to the maximum power requested.
With a two-tone signal, the power indicator gives 50% of the requested power.
Whether for the audio limiters or the ALC there is no limiter. These functions modify the gain to obtain an expected signal level.

The term ALC at 100% may not be correct or may not correspond to the uses of other radios but when we obtain 100% we have an optimum signal that allows to have the requested power.

I can provide a link that explains the phenomenon and provide screenshots and even the gnuradio script.

[image]

sbridger, 2025-02-18:
From the wiring of the sockets I don't think that the earphone output is used. 
Can you check which outputs are actually running the internal speaker and external earphones/ ext spkr?

The earphone is mono (both L&R joined) and wired in push pull. I can make both ear and internal spkr work at the same time by linking across the earphone socket switch. From this I conclude that the internal spkr and earphones are on same circuit.

f5bud, 2025-02-18:
Yes on the "R" revision the headphones and speaker are on the same circuit.
The line out is for USB and bluetooth.

sbridger, 2025-02-18:
Pity. Voice sidetone, and stereo CW are two things it would be nice to have.

sbridger, 2025-02-18:
>But I don't understand where the problem is with the usb or bluetooth and even microphone inputs.
You have to adjust the audio level of the computer or phone so that the ALC indicates 100% and not exceed it.

The issues are:

- the default output from a phone or computer can be 100%/0dB. This overloads the input.
- The optimal level is between -22dB (AuxGain=50) and -34 (auxGain=200). This is counterintuitively low. A user might reasonably expect -10dB which is the overload point.
- There is no indication that it is overloaded. This is the biggest issue. If it was clear when you overload the input, then there would be no problem.

> From there, with a one-tone signal the power indicator is identical to the maximum power requested.

Until overload. Then RF out falls. So the user (reasonably) increases the level...

>With a two-tone signal, the power indicator gives 50% of the requested power.

This is as expected.

I'm not sure that your PC setup is giving you full output level, if you are not seeing the output collapse at -9.8dB.

f5bud, 2025-02-18:
I'm on linux fedora, with 40% output I get 100% ALC. With 100% output there
is no saturation. I don't know how windows handles audio levels.
I can change the display to have an ALC greater than 100% and display
OVERFLOW

sbridger, 2025-02-18:
What is the DC offset value of L2R2 and AuxL?
Could you make this visible in Service mode so we can compare our radios?

So the DC offset would mean that if you increase the WM gain the DC offset also increases, and the available ADC range drops.
Conversely if you reduce the gain the DC offset will be reduced and the ADC range will increase. 
The L2,R2,AuxL can all have gain reduced by up to 12dB (L2/R2/AuxL_2BoostVol)
In theory reducing the gain to -12dB will reduce the offset voltage to 1&fracl;4 

So for the AuxL I suggest trying the -12dB gain setting. That should make the input level more appropriate anyway, and reduce the offset.

For L2/R2 I suggest that the L2/R2_2BoostVol gain reduction should be used as the first -12dB of AGC. 
That would mean that as AGC begins to reduce gain
- the dc offset is reducing and the available ADC range is increasing
- The noise figure is not being impaired by early agc action

f5bud, 2025-02-18:
The gain reduction in the wm8978 will have to be compensated in the
calculations. I need to note the new saturation limits. I will do the test
👍

sbridger, 2025-02-18:
> I'm on linux fedora, with 40% output I get 100% ALC. With 100% output there is no saturation. I don't know how windows handles audio levels.

I suspect that your linux system mixer may be reducing the level gnuradio thinks it is outputting, and 40% is not what the USB DAC is getting.
Single tone overload happens at -9.8dB (I assume that is 32%) in mine

My windows laptop (USB) 100%ALC is hit at -22dB (8%) (AuxGain=50) and my Android phone (BT) 100% ALC is achieved at 18%
On both I can get signal collapse with high levels of a single tone.

Another possible reason for a difference is if my radio has greater offset voltage than yours. 

An offset display in service mode would show that.
If AuxL_2BoostVol was adjustable in service mode I could play with it and see what is happening.

Another possibility is if the OS is changing the output gain of the CM108

sbridger, 2025-02-18:
> The gain reduction in the wm8978 will have to be compensated in the calculations. I need to note the new saturation limits. I will do the test

If its used in the RX AGC, then you will be seeing DC offset steps as it changes too.

I can see these ADC offsets being a variability between radios and an ongoing problem that is likely to come up with someone else in future.
I think you should add to service mode.
- An ADC offset display (perhaps both ADC offset and ADC Max and ADC min)
- AuxL_2BoostVol adjust

sbridger, 2025-02-18:
This is just as a comment for now...
I am thinking that because of the pull down resistors after the AD605, we are losing 1&fracl;2 the RX ADC range, and all ability to use the PGA or the +6dB available in the L2BoostVol.
It is likely that the benefit of removing them would be significant.
That would lead to having two versions of the gain/agc process.
Luckily the large offset voltage would make it easy to detect if a radio had been modified or not.

f5bud, 2025-02-19:
I just made a first measurement on the current version:

The ADC output is 24bits. The max value is +- 0x7FFFFF +- 8388607 decimal

Receive:
register 47, 0b0'0101'0000 // PGABOOSTL: 0db, (I) L2_2BOOSTVOL:   0db, (Digi) AUXL2BOOSTVOL: mute
register 48, 0b0'0101'0000 // PGABOOSTR: 0db, (Q) R2_2BOOSTVOR:   0db, (none) AUXR2BOOSTVOR: mute
DC const mesured: Rx:0x09C808, 641032 decimal

Transmit LSB
register 45, 0b1'0011'0000 // INPPGAMUTEL: on, (mic) INPPGAVOLL: +24db
register 47, muted
register 48, muted
DC const mesured: Tx:0x3095DC, 3184092 decimal

Transmit DIG-L
register 47, 0b0'0000'0101 L2_2BOOSTVOL: // PGABOOSTL: 0db, (I) L2_2BOOSTVOL: mute, (DIGI) AUXL2BOOSTVOL: 0db
register 48, muted, not use in this mode, left signal from usb and bluetooth
DC const mesured: Tx:0x0B230D, 729869 decimal

DF1KW, 2025-02-19:
> Hello,
> thanks for the idea and the link.
> I'll keep the information.
> I don't promise to add this function immediately but I'll do some
> tests and
> see how to adapt internal functions to forth.
> 73, Daniel F5BUD

Dear Daniel,

I hope you are well.

I have been following your intense discussion with "sbridger" on Github
regarding the FX-4CR with great interest.
I can't get enough of thanking you for all the time and effort you put
into this project.

When I read the lines "USB/LSB -6dB at 300Hz, ***@***.***" I felt I had
to write to you again.
I have tried to change the 300Hz low cutoff with some RC elements inside
the microphone, but the results are not impressive.

As I still have the older hardware revision "V", my question would be
whether you intend to implement some of the improvements/features of the
new "R" revision in the "V" revision at some point in the future?

VY 73 de Johannes (DF1KW)

__________/\/*************\/\___________
Dipl.-Ing. (TH) E-Tech  Johannes Koering
Bahnstr. 3 B, 47551 Bedburg-Hau, Germany
Phone: +27-82-4306737 / +49-160-96442627
e-mail ***@***.***
----------------------------------------
https://www.kapstadt.de  Kapstadt Online
https://www.kapstadt.net  Dt. Marktplatz
http://www.suedafrika-forum.org SA-Forum
http://www.cape-town.photos  Fotogalerie
----------------------------------------

f5bud, 2025-02-19:
> Dear Daniel, <br> <br> I hope you are well. <br> <br> I have been
> following your intense discussion with &quot;sbridger&quot; on Github <br>
> regarding the FX-4CR with great interest. <br> I can&#39;t get enough of
> thanking you for all the time and effort you put <br> into this project.
> <br> <br> When I read the lines &quot;USB/LSB -6dB at 300Hz,
> tried to change the 300Hz low cutoff with some RC elements inside <br> the
> microphone, but the results are not impressive. <br> <br> As I still have
> the older hardware revision &quot;V&quot;, my question would be <br>
> whether you intend to implement some of the improvements/features of the
> <br> new &quot;R&quot; revision in the &quot;V&quot; revision at some point
> in the future? <br> <br> VY 73 de Johannes (DF1KW) <br> <br>
> __________/\/*************\/\___________ <br> Dipl.-Ing. (TH) E-Tech
> Johannes Koering <br> Bahnstr. 3 B, 47551 Bedburg-Hau, Germany <br> Phone:
> +27-82-4306737 / +49-160-96442627 <br> e-mail ***@***.*** <br>
> ---------------------------------------- <br> https://www.kapstadt.de
> Kapstadt Online <br> https://www.kapstadt.net Dt. Marktplatz <br>
> http://www.suedafrika-forum.org SA-Forum <br> http://www.cape-town.photos
> Fotogalerie <br> ---------------------------------------- <br> <br> <br>
> <br> Am 08.09.2024 um 19:32 schrieb Daniel Nespoulous: <br> &gt; Hello,
> <br> &gt; thanks for the idea and the link. <br> &gt; I&#39;ll keep the
> information. <br> &gt; I don&#39;t promise to add this function immediately
> but I&#39;ll do some <br> &gt; tests and <br> &gt; see how to adapt
> internal functions to forth. <br> &gt; 73, Daniel F5BUD <br> &gt; <br> &gt;

Hi Johannes,

I don't plan to make major changes to the "V" version, but adding a 300hz
filter is possible. I just have to remember :)

The hardware developments and changes that are on the "R" version would
require a lot of work and time.
For the same reason, the "VR" version will not see the light of day.

73, Daniel

sbridger, 2025-02-19:
> The ADC output is 24bits. The max value is +- 0x7FFFFF +- 8388607 decimal

> Transmit LSB register 45, 0b1'0011'0000 // INPPGAMUTEL: on, (mic) INPPGAVOLL: +24db register 47, muted register 48, muted DC const mesured: Tx:0x3095DC, 3184092 decimal

Well that's a big offset!

> Transmit DIG-L register 47, 0b0'0000'0101 L2_2BOOSTVOL: // PGABOOSTL: 0db, (I) L2_2BOOSTVOL: mute, (DIGI) AUXL2BOOSTVOL: 0db register 48, muted, not use in this mode, l DC const mesured: Rx:0x09C808, 641032 
decimaleft signal from usb and bluetooth DC const mesured: Tx:0x0B230D, 729869 decimal

That doesn't really explain the USB/BT overload issue. 

> Transmit DIG-L register 47, 0b0'0000'0001 // PGABOOSTL: 0db, (I) L2_2BOOSTVOL: mute, (DIGI) AUXL2BOOSTVOL: -12db register 48, muted, not use in this mode, left signal from usb and bluetooth DC const mesured: Tx:0x07F901, 522497 decimal

So I would like to try this (-12dB on USB/BT) and see if it changes the overload issue.

---
RX

> Receive: register 47, 0b0'0101'0000 // PGABOOSTL: 0db, (I) L2_2BOOSTVOL: 0db, Rx:0x09C808, 641032 decimal
> Receive: register 47, 0b0'0001'0000 // PGABOOSTL: 0db, (I) L2_2BOOSTVOL:-12db,
0x05CD99, 380313 decimal

So looking more closely at the datasheet for the WM.
Rin is 4.6k@0dB, 39.1k@-12dB
So as a voltage divider with 100k R to ground, I expect offset numbers less than of 
@0dB 664k   actual  641032
@-12dB 592k actual 380313
So those offsets are broadly in line with what is expected.

We are losing 8% of the ADC range, which is not really a major issue. 
For interest, +6dB gain is 4.3kΩ or 692k offset. So using +6dB would not harm offset. At 50MHz the RX has atrocious sensitivity. I assume that the noise floor comes from the AD605, but it might be worth trying +6dB BOOST_VOL and see if it does make any improvement at 10 or 6m SNR.

I still can't see that those two pulldown R's perform any useful purpose. I'm thinking that removing them would make overloads  more symmetrical, and not require any software change.

So in summary, 
- The L2BOOSTVOL should probably be used to provide the first 12dB gain reduction in the AGC, then the AD605 gain reduces.
- I would like to try -12dB with Aux USB/BT and see if it changes the overload issue.

I'm away for 4 days.

f5bud, 2025-02-19:
I tried to remove these two resistors and the result is catastrophic.
The AD605 has a very low output impedance and it would have to be loaded with a low value resistor.
It would also be necessary to add a capacitor between this resistor and the wm8978.

sbridger, 2025-02-19:
> The AD605 has a very low output impedance and it would have to be loaded with a low value resistor.

Mine has "01D" = 100k. 
I can't read your photo's values.

It has no loading effect on the AD605's 2ohm output.
The only electrical effect is pull down to ground, and slightly changing the high pass freq of the coupling caps. (<3Hz anyway)

> It would also be necessary to add a capacitor between this resistor and the wm8978.

>I tried to remove these two resistors and the result is catastrophic.

There's no obvious reason why. Any idea the nature of "catastrophic?
I wonder if Yu can shed any light on it?

f5bud, 2025-02-19:
On the "V" version the resistor was 47k.
In 2024 I had discussed the DC offset problem by referring to these
resistors.
He had pointed out to me the problem of a huge click on the ADC.
It would be good to connect an analyzer to this signal. It is supposed to
be in the 0hz ... 48khz place.
There must be something else happening here that causes the problem.
Yu noticed this problem and probably looked for a one-size-fits-all
solution :(

On the previous tests I kept the gain reduction at -12db on the aux inputs.
To wait for 100% ALC I have to raise the audio signal from the computer to
80%, I also set the aux-input to 200 in the menu.
This will avoid accidental overload.

For reception, I went back to 0db because increasing the DAC gain also
increases the noise.

f5bud, 2025-02-20:
I made a version with 2 attenuator levels.
The first one reducing the gain to -12db and the second one with an
external attenuator of -22db.
Which gives 0db, -12db, -22db. I didn't cumulate them to -34db, it was too
much

sbridger, 2025-02-21:
(Damn, it's annoying that we have mixed up the RX and TX discussions....)

(V 2.10 20Feb )
TX ALC:
--
BT:fixed
USB: almost, greatly improved.

Having an overload indicator is the only thing left we can do, but I don't know if it is easy to detect overload in the raw ADC values before filtering?

--

Aux gain=100

BT no longer overloads with Ft8TW. 
19% = 90%ALC
21%= 100%ALC
100% = 100%ALC No Overloading.
AuxInLevel 50,100,200 no overloading

USB ifce 
 -17dB = 91% ALC ([was -29dB](https://github.com/f5bud/FX-4CR-V2/issues/31#issuecomment-2652192876) so &Delta; = -12dB)
 -16dB 100%
 -2dB no overload
-1.5 overload begins to appear
 -0.5dB stably overloaded.

*sigh* So close! Good that BT is now problem free, and a big reduction in the chance that a random user will hit this problem. 

The overload characteristic is the same (i.e. it is the same mechanism I think, we havn't surfaced a new one)
Looking on the spec-an at a 500Hz single tone overloading, what we see is a comb of 500Hz. harmonic peaks are equal magnitude. This tells us that the audio is narrow full amplitude spikes, not clipping.

[image]

I think this explains the power drop at overload and 100%: The ALC sees narrow peak-peak spikes and drops in gain right down. Since most of the amplitude is narrow spikes, the average TX power is quite low. The ALC is doing what is expected.

sbridger, 2025-02-21:
(contd)
So the gain change is actually -12dB as expected.
The overload point has moved -9.8db -> 0.5dB i.e. 9dB

If the ADC was overloading (not the amplifier) then the overload point would have moved 12dB

I think this tells us that it is the amplifier input that is overloading. 
Since the amplifier appears to be inverting, this suggests that the overload occurs on the 0V side of the input wave (not the 3V side of the wave), ie the +ve side of the amplifier output.

Being an inverting amplifier (the variable R<sub>in</sub> tells us it is) ,the input voltage doesn't change with signal, until the output saturates. 
The problem is that if it was the output saturating at (say) 2.5V, then the overload point should have moved 12dB not 9dB.

So Vanalog =3.3V , Vmid should be 1.65V , ADC range 1Vrms= Vmid+/-1.41V

I don't understand how that can happen when the gain is -12dB or &frac14; and the CM108 can do max +/-2.5V, and +/-1.41V according to specs.
All I can see is there is some low input voltage where the output inverts. It would be good to have flat radio on the bench to be able to get a scope onto.

If the RX overloads like this it has to be bad news.

sbridger, 2025-06-04:
Hi Daniel,

I have the radio apart trying to improve the rather poor dynamic range (FW 30Apr)

I am measuring the AD605 output voltage,AGC, bias etc
So AGC=2.5V for RF<-58dBm
AGC RF AD605 level
2.5V <-58dBm 35mVrms
1.25V -50dBm 71mVrms
0.67V -40.2dBm  127mVrmsd = 0.36Vp-p
----  -40dBm Surging AGC.

35mV -> 127mV = 11.2dB
-58->-40dB=18dB
The radio shows 0dB ATT at -58dBm, and -7dB attenuation at -40dBm
so 11+7=18.

So the input signal increases 18dB, and the AGC only reduces gain by 7dB. 
At -40dB, 0.36Vp-p I assume the phase reversal overload occurs in the PGA.

Now if I allow the 100k pulldowns to float, or if I connect them to the codec V.mid, then the codec overload happens at -70dB!

--- 

OK, so adjusting the bias levels, I find that phase inversion happens when L2/R2 hits V.mid (1.611V). i.e it appears that the input range is approx 0-1.6V
So to centre bias this I need 15k to gnd, not the 100k I had previously.
With 15k to gnd, I have 777mV bias, and I can run RF to -10dBm, (although the AGC loop is somewhat unstable)

So what is happening with the ATT?
"ATT Fine" seems to change the AGC voltage. L2/R2 bias does not change

When I select ATT -12dB, my bias changes to 354mV (and noise gets quieter)
but when I select ATT -22dB, the bias goes back to 777mV (and the noise gets louder i.e something gets a little noisier, but the signal is reduced)

What is the L2_2BOOSTVOL value? 
Is this being changed for ATT?

sbridger, 2025-06-04:
>On revision "R," the AD-605 is at maximum gain.

No, OUT is connected to FBK = 34dB of gain. Max is 48dB with them not linked

>The WM8978 inputs must be connected via capacitors.

I don't think so. It must not be >V.mid (1.6V) or phase reversal occurs.
I am left thinking that the range is likely to be 0.1 - 1.5V centered on 0.8V

If this is the case, the best solution might be a low value resistive divider off the AD605 with no capacitors, to force 0.8V centre voltage, Perhaps combined with increasing the AD605 gain to 48dB.
If this was done, it would ensure that clipping would occur in the AD605 before phase reversal could occur.

It is difficult for me to work out exactly what is happening because I can't see the ADC values, and I can't stop the AGC from going spastic, and reducing the input level.

It would be very useful to have a test version (or a mode in the where I can:

Disable AGC
See ADC.max ADC.min ADC.mean for each channel.
Adjust the L2_2BOOSTVOL through it's possible values. It would be good if VOL button adjusts L2_2BOOSTVOL and Freq knob adjusts AGC value.

f5bud, 2025-06-05:
The AD-605 is wired to provide a fixed gain of 48dB.

Pin 95 (mRxAtt GPIOB 8) controls the 22dB fixed attenuator.
Pin 29 (GPIOA 5) controls the diode-controlled variable attenuator.

A spectrum analyzer should be connected to the AD-605 output to check the signal quality. There should only be a clean signal limited to 96kHz. Anything above this only causes problems.

I repeat, the WM8978 does not require biasing.
The other inputs do not and work very well.

In a few days, I will upload a test firmware that displays the ADC value and controls the attenuator.

sbridger, 2025-06-05:
>The AD-605 is wired to provide a fixed gain of 48dB.
No it really isn't. FBK is clearly shorted to OUT.

[image]

So if the datasheet is correct, it's 34dB, not 48db.

>  Gain Range
>  FBK short to OUT  −14 to +34
>  FBK open   0 to 48

f5bud, 2025-06-08:
**This is a test version for study purposes only.**

Perform a factory reset first.
The main AGC is controlled by the fine attenuator (0dB, -51dB).
Rf Att: 0dB (Attenuator pin off, L2_2BOOSTVOL = 0dB)
Rf Att: -12dB (Attenuator pin off, L2_2BOOSTVOL = -12dB)
Rf Att: -22dB (Attenuator pin on, L2_2BOOSTVOL = 0dB)

The first line gives the maximum IQ value [512 samples].
The second line gives the same values ​​without the DC component (these are the values ​​used in processing).

The values ​​are in 24-bit signed hex. 0x7Fxxxxxx is a saturation sign.

[FX-4CR-V2-R-Harware.zip](https://github.com/user-attachments/files/20643333/FX-4CR-V2-R-Harware.zip)

f5bud, 2025-06-09:
yes the data in the second line is reversed (copy, paste !) :)

Can you measure the ADC value with the input free, without a capacitor or resistor?

The ADC data is 24-bit signed: 0x0 = 0, 0x7F = -1, 0x1 = +1.
The data displayed is that provided by the DMA transfer without any processing:
```cpp
int32_t adcMaxI = 0L;
int32_t adcMaxQ = 0L;

for (int16_t i = 0; i < SAMPLE_SIZE; i ++)
{
	adcMaxI = std::max(adcMaxI, AdcBuff[i]));
	adcMaxQ = std::max(adcMaxQ, AdcBuff[i]));
}
```
to convert int32_t to float
```cpp
constexpr float32_t Int24ToFloat(int32_t value)
{
	return (value << 8) * (1.0f / (1 << 31));			// ajust sign and convert
}
```

sbridger, 2025-06-09:
>Can you measure the ADC value with the input free, without a capacitor or resistor?

0x009nnn
with noise ranging roughly 0x009000 - 0x009200
That is with no pull down R, and the capacitor grounded (i.e. no noise from the AD605), or to put it anoter way, with C to ground and R is open.
Not sure if that is noisy - seems a bit noisy, but max(512) probably magnifies apparent noise.

The ADC data is 24-bit signed: 0x0 = 0, 0x7F = -1, 0x1 = +1.

16bit 2's complement is:
0x0001 = 1
0x0000 = 0
0xFFFF = -1

so those numbers are 23bit, 2s complement, not 24 bit

sbridger, 2025-06-09:
Perhaps the output mode is wrong. e.g "In I2S mode, the MSB is available on the second rising edge of BCLK" see figs 36,37,38

btw, did you see that I have added a vox wire

f5bud, 2025-06-09:
Yes, I saw it, but I haven't had time to work on it yet.
We'll have to determine when to activate VOX and in which mode.

f5bud, 2025-06-09:
I found what must have been causing the problem.
In the zip file, there's the corrected version for testing (FX-4CR-V2-R).
I took the opportunity to remove the -12dB attenuator.

"FX-4CR-V2-R" is the corrected version.
The second line displays the values ​​in float.

There's still the strong DC component of 0x150000, which represents 16% of the signal.

```cpp
SET_BIT(SAI1_Block_A->CR1, SAI_xCR1_CKSTR);	// 1: Signals change on SCK falling edge
// Signals generated by the SAI change on SCK falling edge, while signals received by the SAI are
// sampled on the SCK rising edge.
```

[FX-4CR-V2-R test + release](https://github.com/user-attachments/files/20657695/FX-4CR-V2-R-Harware.zip)

sbridger, 2025-06-09:
The float I/Q values ate ~0.98 before clipping (-31.7dBm) and go up to ~1.03 after clipping sets in.

I'm not sure what the float values are that they can exceed 1?

sbridger, 2025-06-10:
>The int32 to float conversion is normalized, i.e. '0x7FFF FFFF' = 1.0f
Well that fixes it, but I strongly suspect that the mode or perhaps clock polarity is wrong - I don't see anywhere that the codec can output 23bits of data, so we are losing the LSB of data. (not sure that it makes any difference as there is so much ADC noise)
A corollary would be to ask if the same thing is happening to the DAC data?

>Currently, BOOST VOL uses -12 dB USB + BT for TX.
Perhaps use -6 dB for BT and -12 dB for USB.

BT level a few dB is lower than USB level.
Both are such that they shouldn't clip at 0dB, so you are free to use 0 to -12
I think that 0 was too sensitive when it used that, so -6,-9 or -12.
At -12 for USB, 100% ALC is reached at a PC output volume setting of -10dB on my laptop. That works fine for me, so for me no change is needed, but I really havn't used it much.

f5bud, 2025-06-11:
Yes, there was a small bug. It's now fixed.
See the code below.

I suggest restoring the AGC of the front-end acting on the variable diode pin attenuator and leaving BOOSTVOL at 0dB. This will prevent the AD605 from saturating.

Yu left these resistors in because without them, there was a horrible, very loud noise that saturated the entire radio.
If everything works without them, that's good news.
Can you tell us what the DC component value is?

```cpp
Values ​​provided by the function

// 32 bits integer: (For reference)
	Int32(0x7F'FF'FF'FF): +2147483647
	Int32(0x00'00'00'00): +0
	Int32(0x80'00'00'00): -2147483648

// 24 bits integer:
	Int24ToFloat(0x7F'FF'FF): +0.9999999
	Int24ToFloat(0x00'00'00): +0.0000000
	Int24ToFloat(0x80'00'00): -1.0000000

	FloatToInt24(+0.9999999): 7F'FF'FF
	FloatToInt24(+0.0000000): 00'00'00
	FloatToInt24(-1.0000000): 80'00'00

// 8 bits integer:
	Int8ToFloat(0x7F): +0.9921875
	Int8ToFloat(0x00): +0.0000000
	Int8ToFloat(0x80): -1.0000000

	FloatToInt8(+0.9999999): 7F
	FloatToInt8(+0.0000000): 00
	FloatToInt8(-1.0000000): 80
```

f5bud, 2025-06-11:
I've reintroduced the variable input attenuator (First AGC).

L2_2BOOSTVOL and R2_2BOOSTVOL at +6dB
Added LIMBOOST at +3dB DAC volume boost.

When a signal is in the ADC window [48kHz] (First AGC) it kicks in at -60dB.

- The attenuator marked fine acts as an attenuation send.
- The indicator (top left) shows the attenuation of (First AGC).
- The RSSI calibration needs to be redone.

Can you test this version and give me your feedback?

[FX-4CR-V2-R.zip](https://github.com/user-attachments/files/20694284/FX-4CR-V2-R.zip)

sbridger, 2025-06-12:
>Yu left these resistors in because without them, there was a horrible, very loud noise that saturated the entire radio.

Well yes - that was caused by the numeric fault. It has nothing to do with the circuitry, which works properly without the R.

>I suggest restoring the AGC of the front-end acting on the variable diode pin attenuator and leaving BOOSTVOL at 0dB. This will prevent the AD605 from saturating.

It is to be expected that the IF will saturate because it is broadband.

If you try to use AGC based on the whole IF baseband to reduce gain, then you must lose sensitivity for in-channel weak signals, when there are multiple strong signals in the IF. Thus the radios dynamic range is reduced. 
Signals add by V, so if you have 5 off-channel signals at S9 (-73dBm) the pk is -43dBm. 
If you had a -60dBm off-channel agc threshold, you would start to desensitise the in-channel at only -90dBm ~S6.

So you want to leave reducing gain until overloading is going to create artifacts. And you want to try and reduce overload artifacts as much as you can, so you can allow as much overloading as you can.

f5bud, 2025-06-12:
I chose -60dB because peaks start to appear on the FFT above this point.
I made a correction regarding the transmit attenuator. It's on the same line and wasn't set to zero on transmit. The zip code will be in another post !
I'll add a menu to block the first AGC.
The low line of the FFT is a new feature requested by an OM.

sbridger, 2025-06-12:
So the AGC kicks in at -39.1dBm on-channel,  -42@ 14.088 (i.e. -12kHz is 3dB above on channel, this is the IF filtering)
It is not affected by RSSI calibrate as expected.

---

OK, so to look at the rising baseline. Is it from the generator or the radio?

CW BW is 800Hz = 29dBHz. (just assuming 800Hz is the noise BW)
My R&S generator is speccd at <-100dBc/Hz@>=20kHz which = -71dBc@800Hz @F.rf=14.08MHz

So currently 
RF  RSSI
off  -129
-
-42.6dBm  -119   excess noise 10db greater than internal noise.

So excess noise is -42 -119 = -76dBc
This is somewhat better than the R&S spec of -71

My new IF filter rolls off below 7kHz, so if I set F.IF to 1kHz, the IF signal hitting the ADC will be significantly reduced.
Thus if the noise stays the same, it is phase noise from the signal generator or the SI5351/buffers etc
If the noise drops, then it is a function of the IF signal level on the ADC, software or AD605.

F.IF Noise Offset RF  RSSI
-- off -129
4kHz 20kHz -42.6dBm -119
1kHz 23kHz -42.6dBm -119
DC 24kHz -42.6dBm -119

---

Conclusion: The rise in baseline noise is from the RF, not an artifact of the IF/ADC

If I want to separate the source as being the si5351 or the signal generator I need a low PN source. 
This is where I regret not getting a whole bunch of different frequency analog crystal TCXO's when I worked at Rakon!

sbridger, 2025-06-13:
I cut the tracks to bring the AD605 gain up to 48 dB.

Now 10dB SINAD 14.1,USB,2.7k BW, BOOSTVOL -3dB is -121dBm  
Not sure if I will keep it like this.

It does show up that the RSSI calibration moves the spectrum graph bottom line, as I have moved it by 12dB, and the smps noise got lost.

sbridger, 2025-06-13:
Food for thought:
(i.e. not a request)

AGC could be the sum of 3 mechanisms.

_Wide or IF AGC_ - Acts to reduce RF (PIN) when the whole raw ADC approaches clipping (e.g. -40)
_Narrow or Channel AGC._ First acts digitally, then reduces RF (PIN) once SNR is sufficient e.g 30-40dB. This maximises dynamic range, by reducing gains as soon as the channel has enough signal to be 30-40 db above noise floor
The IF (AD605 and/or X5R caps) appears to generate harmonics and IMD products from -65dBm. The threshold for this might be -80 say in the -90 , -60dBm range.
_Floor AGC_: Looks at the lowest bins in the spectrum scope, and reduces RF gain. E.g. if the lowest bin is 20dB above the noise floor, then you can have -17dB PIN all the time. This is what the manual RF gain control does, so is really a way to automate that.

sbridger, 2025-06-18:
> With the correction, we achieved a signal twice as strong (+3 dB) and much better resistance to saturation.

The big problem was that the effect of it was to shut the AGC down hard. That turned saturation/clipping into the various farting noises and the radio being unable to rx at all.

> I was able to enable the wm8978's high-pass filter.

An effect will be that the max/min and clipping values will shift. i.e. when the ADC clips it won't be 0x80.. /0x7F.. anymore. 
BTW, probably use the highest HPF freq. (I have my analog HPF before the AD605 at~7kHz now)

> The resistors can be removed later.
Indeed. I don't think there is big advantage to remove them at BoostVol 0 to +6 . 
I am making some changes to the IF filter components, perhaps in a few weeks I will have settled on values.
I would like to have a recommended mods list in a month.

f5bud, 2025-06-19:
I just added an "OVERRUNS" message right on the spectrum when the signal is too strong (-30dB on my hardware). The threshold is set to 0.5 of the maximum value, which is 1.0.
This message can only appear when the AGC is turned off.

I'll update the repository as there are several changes and corrections.

sbridger, 2025-06-23:
Well the previous measurement showed the FX4 was as good or better than my R&S siggen.
I popped in to the university, which had an HP8640b when I worked there 40+ years ago, and was which still ditting neglected in a store room, and more amazingly, still works.

CW BW is 800Hz = 29dBHz. (just assuming 800Hz is the noise BW)

Due to the low noise, it had to be measured with DC IF frequency i.e. measuring the 24kHz offset noise
So currently

F RF RSSI 
14.310 off -133
14.310 - 20 -118   
50.1  -20 -120 (= -117 )

14.1MHz -147dBc/&sqrt;Hz
50.1MHz -146dBc/&sqrt;Hz

This unexpectedly good.

This above the expected noise floor of the generator at ~-155 from what I could find, so probably the FX4 performance, though I did expect 50MHz to be 6dB/octave worse.