f5bud/FX-4CR-V2 #33: PWR (ALC) sensitive to voltage and temperature

sbridger, 2025-02-09:
So I think (by observation) that the power output is predictively set, rather than being set by feedback from the forward power sense diode.
i.e. a certain DAC value set in calibration is predicted to give the desired output.
Testing at 14.1MHz, CW
 dBm@  5W   10W  20W
10.2V  37.2 40.3 41.8
12.1V 37.6 40.8 43
14.5V 38.1 41.4 44.1

Keeping key down at 5W,13V, power rises 0.3dB as temperature rises.
In linear terms 
Set 10W
10.7W
12W
13.8W
 
At 14.5V, 20W, the power out when hot will be ~28W
At this high end the much higher heat is risking the finals.

My feeling is that ALC control should be based off the the forward power diode. This also makes it less sensitive to calibration. However Yu has used an swr bridge design which is common, but wrong. The use of a single balun core when two separate transformers is correct, may result in the reflected power crosstalk to forward power, making it unreliable, and would need checking. (it also results in poor swr measurement, but that doesn't really matter much)

priority: working at the moment, but should be investigated.

f5bud, 2025-02-10:
Using power detection is not very stable.
It is difficult to rely on this value.
The radio must be calibrated with the voltage intended for use.
A possible evolution is to make several points per band.
For example 5 points from 7.0mkz to 7.3mhz. The output filters are far from
flat.
This implies an automatic calibration and reducing the number of memories.
The eprom is only 2kb.

f5bud, 2025-02-10:
> In what way is in unstable?
> This is exactly what (analog) radios use.
> They do have far faster attack time than the FX4 has had.
> Icom and Yaesu radios do use the Collins SWR/forward power detector
> arrangement, and not the single balun core. I don't know whether the effect
> of that is to increase or decrease the apparent forward power at high swr.

The HF detector is connected to an input of the stm32 which is programmed
in adc. The reading is not stable yet there is a whole filtering logic.

sbridger, 2025-02-10:
I'll have a look at what schematic I have, later. Off to an island to install solar panels now...

sbridger, 2025-02-11:
>The HF detector is connected to an input of the stm32 which is programmed
in adc. The reading is not stable yet there is a whole filtering logic.

Looking at the 23Jan2022 schematic, 
Time constant of the detectors (C47,R24) is 33Hz
Opamp input filter (R17,C48) 1.7kHz
ADC input filter (C?,R28//R29) = 3.33kHz
Circuit subtracts the diode drop in hardware.
VADC = 1/2 Vdiode. i.e. gain is 1/2

So there is no reason for the design to have noisy FWD/REFL detector values, when a CW carrier is output,  from the schematic point of view. It is well filtered, and there is allegedly a cap at the CPU input.

(Also, the design seems to heavily filter the 48kHz ripple that could be used for TX image measurement. So auto tune TX based on this technique is probably out)

I wonder if the instability is therefore coming from the baseband drive to the TX. However I didn't observe excess amplitude (envelope) noise on the CW.

How many counts of instability / noise is there?

sbridger, 2025-06-25:
>The ADC measurements for Volt, Pwr, Swr, and Temp are quite noisy despite using the filters integrated into the stm32h750.
I use the internal voltage reference because the external reference is even noisier.

Usually the internal vref is presented to the avref pin, so it can be bypassed. There is no bypass on this pin, but fairly easy to add one. What is the internal Vref voltage? There is also the possibility of it fighting with the external ref if it's higher.

f5bud, 2025-06-26:
The reference voltage is 2.5V.
The internal layout of the SMT32H750 isn't straightforward.
There's no indication that the internal reference can be output via the Vref pin.

I tried changing VREFINT but the final result didn't change.
```cpp
SET_BIT(ADC12_COMMON->CCR, ADC_CCR_VREFEN); // 1: VREFINT channel enabled
```

By changing VREFBUF, the DAC outputs were no longer outputting anything.

I took the opportunity to measure the consumption:
|Mode|Consumption|
|--|--|
|Power off|1mA|
|Dfu mode|187mA|
|Rx, speaker at 0|228mA|
|Tx no signal|690mA|

sbridger, 2025-06-29:
Section 6.16 pg 93 Fig 11
shows the Vref connected to the external pin so it filters the internal reference, and the footnote requires a bypass capacitor

[image]

f5bud, 2025-07-03:
I don't think we'll see any improvement as things stand. I get (31845-31817)/31845=10.7 mV

Yu could have avoided the external reference (resistor + TL431) and added a decoupling capacitor. The ground planes should be improved.

Regarding power consumption, the processor operates at 480 MHz and the overall current is 0.228 mA. Down to 300 MHz, the current is 0.210 mA, a reduction of 219 mW.
At this frequency, the actual processing IQ at 96 kHz requires 25% of the CPU resources.

I'll leave this clock frequency at 300 MHz for now and resume my tests on the noise reducer using spectral analysis and see if the CPU time is correct.

[Test-300M-25-07-03.zip](https://github.com/user-attachments/files/21034089/Test-300M-25-07-03.zip)

sbridger, 2025-07-08:
> I'll leave this clock frequency at 300 MHz for now and resume my tests on the noise reducer using spectral analysis and see if the CPU time is correct.
> 
> [Test-300M-25-07-03.zip](https://github.com/user-attachments/files/21034089/Test-300M-25-07-03.zip)

I tried this, but it didn't use the internal Vref, and I have disabled mine, so I didn't test further...

sbridger, 2025-07-09:
>In this version, if there's no voltage present on the VREF pin, I enable the internal reference. (It should work.)

Good solution.

>This is also a first test of the noise reduction using spectral analysis.
>The CPU is still running at 300 MHz.

I will post comments in the respective topics.

sbridger, 2025-07-13:
If all goes well, I'll publish it.
[FX-4CR-V2-R-25-07-09.zip](https://github.com/user-attachments/files/21148910/FX-4CR-V2-R-25-07-09.zip)

1) At power on the screen flickers, especially in the top blue status line. Not a problem as such, but might indicate a timing issue?

2) ATT operation: If FirstAGC=OFF, then only coarse ATT shows. If FirstAGC=ON, coarse and fine ATT shows. If you intended to have two different behaviours, I would expect them to be the other way around. 

3) It seems to me that the power meter peak capture has changed. My test set shows 42dB peak, but the meter is only showing ~7W. The meter is correct on CW, so this appears to be a peak hold issue. Perhaps you have increased the averaging time of the adc inputs before they reach the metering routines?

sbridger, 2025-07-15:
The RIT function menu item has disappeared also. So I can't set RIT to PTT

f5bud, 2025-07-15:
The RIT, XIT, PTT function can be activated by a long press on TUNE.