f5bud/FX-4CR-V2 #35: Poor Sideband Suppression. No microphone HPF.

sbridger, 2025-02-17:
While I was looking into the audio overload/ALC issue, I found that the sideband suppression is poor for F<sub>audio</sub> <500Hz (i.e. LSB, not image (-48kHz) suppression).

This is 14.1MHz, Dig-USB, audio from USB, Aux-in-gain=50.
Suppression is not sensitive to audio level. Level is set just above 100% ALC

I expect that "good" suppression is -50dB. Acceptable is -43dB 
100Hz -9.1dB
300Hz -24.7dB
500Hz -36.1dB
1000  -68dB
Dig-LSB, USB, LSB is the same.

There appears to be no highpass filtering of the audio, or preemphasis.

I have tested the mic input, and it is the same, no apparent HPF.

What is the SSB generation technique used? (the audio->24kHz IF signal, not the 24kHz->RF mixing)

Carrier suppression is -55dB@F-24kHz.

Expected: 

Sideband suppression > -43dB@300Hz
An HPF in the microphone path @300Hz for USB/LSB

If Dig-USB does not have an HPF (which is reasonable), then adequate sideband suppression should extend down to 100Hz or lower.

sbridger, 2025-02-17:
<small>Carrier suppression meets [ETSI301-783](https://www.etsi.org/deliver/etsi_en/301700_301799/301783/02.01.00_20/en_301783v020100a.pdf) Table 6.
However it might be good practice to switch sides to keep the spurious carrier within the amateur band. (priority: very low)</small>

f5bud, 2025-02-17:
Description of audio signal processing:

The raw signal from the 24-bit ADC converter is then normalised to a 32-bit
float.
It first passes through a 50 Hz high-pass filter to remove the DC component.
It is then limited to -1 or +1 in amplitude.
For LSB and USB modes, the signal is first mixed with a BFO of (2,700 Hz /
2), then filtered at (2,700 Hz / 2), and finally mixed with an OSC of (24
kHz - 2,700 Hz / 2).
The whole system operates in complex IQ. This is a Weaver modulator.
The LSB uses sine, and the USB uses cosine for the BFO.
If the IQ balance is correct, the residual signal at 48 kHz becomes zero.
At the output, we obtain an LSB or USB signal set to 24 kHz.
The hardware mix of the fc - 24 kHz VFO and the signal gives us the
transmit signal.

Source of inspiration:
https://forum.pjrc.com/index.php?attachments/weaverdocument-pdf.9746

f5bud, 2025-02-17:
> If the audio BW is 300-2700Hz, then the weaver LPFs should cutoff at
> (2700-300)/2=1200Hz . The weaver osc should be at 300+1200Hz=1500Hz
> If you change the highest or lowest audio frequency, the the filter LPF
> and the weaver osc freq all change too.
> For 300-2900Hz, Fc=1600, FLPF=1300Hz
>
> I think what is missing in the frequency plan is any guard band (see
> figure on pg 6), thus the other sideband is only just off the filter, when
> it needs to be well down the slope.
>
> (from memory - it's late)
>
> The Mic input should have a 300Hz HPF, as external noise (wind, machinery)
> can have a lot of amplitude below 300Hz. The mic filtering is adding to the
> sideband filtering on TX.
> ------------------------------
>
> Mic clipping
>
> If the audio input is going to be significantly clipped, then it should
> have preemphasis first.
> It should ideally be LPFd as a whole lot of high freq noise will IMD in
> the clipper, back down into the 3kHz passband.
> Using a soft clip curve instead of a numeric hard clip reduces the audio
> effects too.
>
> Analog clipping in radios didn't want to add a whole lot of audio
> filtering, but in broadcast audio you clip and compress separately on
> octaves. That means that the 3rd harmonics produced by clipping don't fall
> in the post clip passband and get removed.
> In DSP the filters are free-ish, so I suspect you would ideally break mic
> input into two bands around the voice deadspot (1100Hz) ie 300-1100 and
> 1100-2700Hz. Each of those would be clipped/compressed, and LPF'd, then
> combined.
>
> A microphone

Yyes you are right about the guard frequency. I made the change.

The limiter does not clip. It reduces the gain, there is no distortion.

sbridger, 2025-02-17:
>yes you are right about the guard frequency. I made the change.

I'll test it when it appears. Since it's a recal, I won't try the 15Feb version.

Do you use the WM8978 ADC HPF for (or to supplement) the 300Hz? (datasheet fig54)
What ADC sample rate do you use for mic/aux input?

The 5band equaliser could be used to achieve pre-emphasis or to allow voice/mic tuning. I notice that this is common on newer radios, I wonder if they have this feature because the mic chip has the equaliser built in?

f5bud, 2025-02-17:
The WM8978 is overclocked at 96khz. I don't know if the equalizer works. I
can give it a try. All filters are 7th order by-quad. In reception the ADC
input comes from an AD-605. In transmission the input switch selects the
mic input or the aux input.

sbridger, 2025-02-19:
V17 Feb
Fixed.

USB/LSB   -6dB at 300Hz, -16dB@200Hz
Sideband suppression >50dB at all freqs. (single tone)
Performance: excellent.
---

I'm not sure if DIG-SSB should (eventually) have some of the same.

What is the current F<sub>LPF</sub> and F<sub>weaver</sub> in DIG-SSB?

EN300-373 8.9.3 appears to set a limit of 31dB @ -100Hz and AS4770 -25dB , but this is for the 2 tone IMD test products.
Neither imposes any sideband suppression requirement below 300Hz

Action: None. Avoid source energy below e.g. 100Hz

f5bud, 2025-02-20:
> What is the current FLPF and Fweaver in DIG-SSB?

For SSB I limited to 300..2700 and the rest 30..3000. For DIG I can limit
to 100..3000

sbridger, 2025-02-20:
100Hz would definitely be better I think. 
200 would be my pick for the reasons below...

Using the spec an, there is 1&frasl;f energy between 30-100Hz.
The reason to have TX af filter below the nominal 300Hz, is so the filters are flat and not especially distorted at 300Hz. It is better to increase digital mode BW by going higher than by putting any energy close to the carrier.

100Hz, would also somewhat reduce 50/60Hz earth loop noise. However a lot of earth loop noise is actually 100/120Hz. 

I guess 200Hz minimum would be better for that. 
However since we are using direct USB connection there really shouldn't be any hum.

What is RX?
I assume that the RX is the same i.e. having F<sub>min too low on RX results in poor RX sideband suppression and extra noise too.
So the RX should be more like 100 or 200Hz

So:
100Hz definitely a better choice for TX
200Hz prob better for RX and good for TX.

f5bud, 2025-02-25:
I adopted 200 hz as low frequency for Rx and Tx.
I made a small distinction for SSB with 2700 hz for high frequencies.
I removed a filter that had no place there. This allowed me to gain a few microseconds in the processing.
NR should no longer block. There is only 5.33 ms to process the signal (512 samples / 96000).

closed 2025-08-31 by sbridger