f5bud/FX-4CR-V2 #37: RX goes silent after overload

sbridger, 2025-02-21:
While testing overloading with the signal generator, I found that after I drive the radio into in-band overload, the  audio diasappears. (CW, 14.1MHz, Gen=14.1MHz, RF > -40dBm)

It does not return when the level is reduced, only if ATT is pressed to change values. 
S meter is still changing, just the audio has gone.

It overloads (and goes silent) at

Att 30, Overload at -40dBm
18   ovl -36dBm
8 ovl -17dBm

When radio overloaded out of band it starts clicking at -36dBm, but it does not go silent.

This may have be an existing issue, I think I have had the radio go silent in the field, and had to restart it.

f5bud, 2025-02-21:
This should not be a big problem. A division by zero is not excluded in
this situation. I will look in a few days

sbridger, 2025-02-21:
It might also be something that came in with the NR.
AGC time seems far longer than before 17Feb. I recall that this happened before and made the rx seem deaf when it was actually something being set to a very big number...
NR crashes immediately. The first couple of times NR waited a second before crashing

f5bud, 2025-02-21:
In the presence of a very strong signal, the gain is so reduced that the
system takes a very long time to recover.

f5bud, 2025-02-24:
The NR crash is because the delay value "D:0" is a 16-bit int while before it was another 8-bit char variable.

I should have informed you that a factory reset was required.
The configuration data is in a structure and any change causes improbable effects.

In the radio code, there is a watchdog that triggers if a part of the program crashes.
That's why sometimes the radio stops without warning. It's a safety feature!

Another point, the value "F:1" works backwards, 1 is the strongest and 8 the weakest. In the next update this will be fixed.

max-arnold, 2025-02-25:
>  The configuration data is in a structure and any change causes improbable effects.

This problem could be solved by adding a version number to the configuration data area and adding forward (and possibly backward) migrations. The downside is that all these migrations will need to be kept forever.

closed 2025-07-26 by sbridger