f5bud/FX-4CR-V2 #39: Rx overload

TCB911, 2025-02-26:
I just loaded the new firmware. I find I have to enable the attenuator or I get a "tic"  and the receiver is easily  overloaded. It is better than the very last version.  Hopefully my explanation makes sense...  It just keeps getting better.  Please let me know if I should be running the attenuator all the time or 
 do I need to recalibrate.    
Thanks  de VE3IQL

f5bud, 2025-02-26:
Thank you for your report.
The overload is not related to the calibration.
When did you receive the radio?
This is to check that there is no hardware change.
Daniel

f5bud, 2025-02-26:
I added a menu in IF.ATT to reduce the gain at the ADC input

TCB911, 2025-02-26:
As well Daniel, with the IT.ATT menu I find the receive audio is better...

Terry

f5bud, 2025-02-26:
I think there is a hardware change. I just discovered that the gain control does not work as we thought. The new menu will probably allow to pass

TCB911, 2025-02-26:
That might be something to ask Yu.  The problem occurred when the bands were really busy and of course now as I try to find signals close together, the bands are pretty empty.    I'll try later but your menu seemed to calm the situation nicely..

f5bud, 2025-02-26:
That's what I did :)

TCB911, 2025-02-26:
Daniel, I found some crowded spots on 40meters and I am having no issues.
Problem solved.

One other question. I operate mostly SSB and was wondering if more memory
can be set aside for voice recording space.

I  REALLY like the AM/FM button TX feature for an antenna tuner. Brilliant!

Thanks, Daniel,   you have really made this into a fun little radio
Terry de VE3IQL

sbridger, 2025-02-27:
>One other question. I operate mostly SSB and was wondering if more memory can be set aside for voice recording space.

I noticed that the CM8978 can do &micro;/Alaw direct companding digitise. This might be a way to reduce the storage reqs without any actual code...

f5bud, 2025-02-28:
By analyzing and probing the main PCB, I noticed that the AD605 is wired in fixed gain of 48 db. The FBK pins are not wired and the VGN1 and VGN2 pins are connected to VREF.
I do not know by which circuit Yu varies the HF gain.
I started with the hypothesis of a variable gain from 3db to 48db, but it is probably wrong.
I contacted Yu for explanations on this part, but I do not have an answer for now.

TCB911, 2025-02-28:
> The wm8978 could be used to record voice, but to make calls it would be
> necessary to decode the compressed voice by program.
> Currently the voice is decimated to 6000 Hz (96000 / 16) - 8 bits and
> recorded in the stm32 flash memory (like the firmware).
> There is no other memory to store data. STMicroelectronics guarantees
> 10.000 write cycles !
>
> Increasing the recording time requires more Ram, what would be the desired
> recording time?

Hi Daniel
Even another 5 seconds would be a bonus. We are VERY happy with what you
have achieved so don't get me wrong. This has added a really nice feature
to the rig.
This is a fascinating project.
The added IF.ATT  menu has helped tremendously. It has reduced issues with
strong adjacent stations and the RX  audio quality is much improved as I
had mentioned.
TX audio reports are quite satisfactory as well..

f5bud, 2025-03-03:
Thank you for your request.
I realized that I was limiting the area too much.
I could add a few more seconds but there wouldn't be enough memory for other functions.
The maximum length of a voice message is 15 seconds.

f5bud, 2025-03-13:
I have an update from Yu on the attenuator.
The "R" revision platforms use a pin diode attenuator ref: BAP64Q.
The AD605 is wired for a gain of 48dB.
The pin diode attenuation ranges from -60dB to -3dB for a control voltage of 0V to 3V.

f5bud, 2025-03-14:
For your information, I have plotted the curve of the pin diode attenuator.

[image]