f5bud/FX-4CR-V2 #43: RX only mode

max-arnold, 2025-03-25:
The documentation mentions the RX only mode, but I found no way to activate it (sorry in advance if I missed it in the docs).

At home I have an RX-only external wideband antenna with an amplifier. Hitting PTT could be disastrous both for the amplifier and the radio. Also the radio could be used with a short untuned telescopic whip antenna while traveling. The power consumption and the dimensions are quite small.

My suggestions for the RX-only mode:

* No concept of bands (just select the frequency you need from the full range)
* Arbitrary modes (AM/FM/LSB/USB/CW)
* Appropriate bandwidth choices
* ATT should work
* MEMORY should be supported (to save specific stations, as well as to store jump points to specific bands)
* All ways to trigger the TX should be disabled in this mode (PTT button, Mic PTT, CW key, Tune button, VOX, USB, etc.)

dc1rdb, 2025-03-25:
As far as I remember, RX only is being activated by long press AM FM. Don't have the rig with me right now, though.

f5bud, 2025-03-25:
Okay, I'll turn this function back on. The radio will work as before, only transmission will be locked.
The AM.FM button is the tune mode.
The M.V button will activate the Rx-only mode.
"SWL" will appear in the top left corner of the screen.

max-arnold, 2025-03-25:
A minor inconsistency in the docs:

[image]

f5bud, 2025-03-25:
Yes 🤪

closed 2025-03-26 by max-arnold