f5bud/FX-4CR-V2 #44: VOX support

max-arnold, 2025-03-28:
Are there any chances to see VOX support in the V2 firmware?

The use case is often asked in various FX-4CR groups: FT8 transmit support in iFTx on iOS via bluetooth (for portable use). 

It looks like iOS doesn't allow direct access to serial-over-bluetooth, so triggering TX is only possible by monitoring the incoming audio level.

For now iFTx over bluetooth works in RX mode only.

max-arnold, 2025-03-28:
By the way, I was unable to make WSJT-X work over bluetooth on macOS either. The serial-over-bt port does appear as `/dev/tty.FX-4CR` and `/dev/cu.FX-4CR`, but `rigctl` times out when connecting to it (tried different port settings with no success).

f5bud, 2025-03-28:
Hello Max,
Technically, there's no way to do a VOX.
On reception, the ADC is connected to the RF mixer and the microphone input
is disabled.
Bluetooth connections work well on phones and tablets. Under Windows, it
seems possible. I haven't had success with Linux, and what about macOS ???

Under Linux, it's /dev/ttyUSB0,  /dev/ttyUSB1, etc.

73 Daniel

max-arnold, 2025-03-28:
> Technically, there's no way to do a VOX.

Sorry, I mean audio-input-over-bluetooth activated TX mode (and not via the mic). The FX-4CR succesfully connects as a soundcard to my iPhone, and the iFTx app is able to decode FT8. It also transmits the audio over BT back to FX-4CR (I managed to make a couple of QSO by pressing PTT manually). The missing part is a way to trigger the TX mode when there is some audio signal coming in over bluetooth.

On a macOS machine pairing the transceiver over BT adds two tty ports: `/dev/tty.FX-4CR` and `/dev/cu.FX-4CR`. But the following command times out (and I tried different port settings):

```
rigctl-wsjtx -m 2031 -r /dev/tty.FX-4CR -s 115200 -vvvvv -Z f m v
```

f5bud, 2025-03-28:
There is a CAT command that switches to transmission

max-arnold, 2025-03-29:
Also a quick unrelated question: does the FX-4CR hardware support bluetooth headsets? Or it can only act as a peripheral?

f5bud, 2025-03-29:
I have no experience with bluetooth headsets, sorry!

sbridger, 2025-05-22:
>Technically, there's no way to do a VOX.

With bluetooth. With USB it should be possible. Perhaps you should try a USB cable to your iphone. If the usb audio works, then perhaps a VOX would be worth implementing.

sbridger, 2025-06-03:
I am wondering if we might be able to add a wire on the cpu/adc board to link the USB/BT audio-to-radio to a spare cpu ADC pin, which could be used for VOX detection.

Is the  USB/BT audio-to-radio signal on the cpu bd?
Do we have a spare ADC pin on the cpu with a pad?

max-arnold, 2025-06-04:
My assumption was that the BT is connected digitally to the CPU so it can always receive the auidio stream over BT without any ADC and independently from the TX state.

sbridger, 2025-06-04:
BT / USB / MIC are connected to the WM8978 codec chip, not the cpu.
This chip selects which channel(s) are connected to the ADC. 
In RX, both ADC are committed to I/Q rx signals. There is no spare ADC in the codec chip.

max-arnold, 2025-06-04:
Does it mean there is an additional digital -> analog -> digital conversion for the bluetooth audio?

What chip or BT module is used in FX-4CR?

sbridger, 2025-06-04:
>Does it mean there is an additional digital -> analog -> digital conversion for the bluetooth audio?
Yes.
The CPU does not do USB or BT. 
There is a USB-Serial and USB-AnalogAudio  and BT-Serial+AnalogAudio modules

sbridger, 2025-06-05:
As a note, if a link wire has to be added to the CPU, it is far easier to solder it between two pads. i.e connecting to pin 15 and pin 16. So pin 15 could be used, pin 16 should be arranged so it can also be shorted to pi15 with no issues.
I will measure the voltages on the BT+USB signal tomorrow.

sbridger, 2025-06-07:
So I found it was easy enough to add a link wire to pin 15.

[image]

Factory Vbias=0.96V. Signal range at USB 0dB= 0.2V - 1.65V . This causes sign reversal, so I have corrected Vbias by adding a 200k resistor parallel with factory 100k = 60k Total. 
Corrected Vbias=0.8V

[image]

f5bud, 2025-06-13:
I'll test it out sometime when the other changes are stable.
Pin 15 is already in the code because the "V" version was supposed to use it. There was nothing behind it, maybe on the FC-4C.

I also need to solder a wire!

sbridger, 2025-06-13:
As I said, it's a lot easier when the wire sits between 15,16 and bridges them. Getting it to perch on top of the pad without shorting adjacent pads is tricky even under the microscope. Maybe I need to lay of the sauce.
So pin 16 should be a free input if possible.

I can test it when you have something.

sbridger, 2025-08-20:
This is also a problem in Sailfish where we have to use android apps. Sound is routed from Android, but special BT interfaces like serial are not. 
I have also noticed that (on android phone) while the audio from ft8tw is quite reliable, the cat control is far less stable. VOX is a basic reliable fallback to make it work.

>It could be used with a Bluetooth microphone, and the audio routing would need to be changed using a menu to choose between digital and microphone.

Can you pair a BT mic to the FX4?
I think voice/mic vox is a different problem  (e.g. voice vox works better with a delay line), and not what has been requested.

f5bud, 2025-09-24:
Timer 4 is used for AF encoder pin 60, 61

f5bud, 2025-11-11:
Yu needs to send me some cards.
I'll link Simon's WM8978 to STM to see if it's possible to do anything.