f5bud/FX-4CR-V2 #45: Calibration

TheDireHavok, 2025-05-06:
Is it normal that when I try to calibrate wattmeter external/internal, that it maxes out? I.E. I am at tx pwr for ext calibration and tx pwr is at 200 and ext wattmeter displays 11-12 watts and output is set at 10watts. And internal calibration I can only get it down to 10.53 watts and is maxed out at 1300.

f5bud, 2025-05-06:
On the strip, the first value is a coefficient and the second is the power in watts.
This is the coefficient you will note to record the calibration values.

TheDireHavok, 2025-05-06:
Just to clearify, this is what I am having issue with on my rig. So now if I set 10 watts for output power on rig, meter is higher by 1 watt due to inability to calibrate any more. (In Bold is section I am talking about)

12.3 - Wattmeter calibration
 📌 80m band (3.600 MHz), CW mode, Power HF 10.0W
 (You can measure power with your tinySA, 10 W = 40dBm.)
 First step:
 1. Connect power meter with RF dummy load or attenuators and tinySA.
 2. Press 
MENU and turn 
3. Press 
TUNE to select "80m-TxPwr".
 PTT to transmit and turn AF to set 10.0W on your external power meter.  **This is part where I can't turn AF any further to show 10watts on ext wattmeter, lowest value is 11**
 
Second step, (internal power meter): **Was able to get this calibrated but may be off due to 1st step**
 1. Turn 
TUNE to select "Watt-Cal".
 2. Press 
PTT to transmit and turn 
AF to see 10.0W on the "Watt-Cal" line.
 (This setting acts as a divider. The indicator increases as the value decreases)

f5bud, 2025-05-06:
First step, (**external power meter**):
1. Turn TUNE to select "80m-TxPwr".
2. Press PTT to transmit and turn AF to set 10.0 W on your external power meter.

Second step, (**internal power meter**):
1. Turn TUNE to select "Watt-Cal".
2. Press PTT to transmit and turn AF to see 10.0 W on the "Watt-Cal" line.

Can you get 10 W at the first step on your wattmeter?
If yes, proceed to the second step which consists of calibrating the internal power meter.

All the values ​​are on the green banner.
You must stay in the menu to complete the procedure.
I don't really see what's bothering you.