f5bud/FX-4CR-V2 #50: Feature request: Syllabic Squelch

sbridger, 2025-06-23:
I would find effective SSB voice squelch, e.g. syllabic squelch quite useful - more that noise reduction.
(My experience of noise reduction is that it reduces noise on strong signals, but weak signal intelligibility does not actually improve)

DSP gives one big advantage over trying to make reliable analog squelch - you can put a long delay line in so that the voice detect/squelch has a longer decision time. 

You can also use squelch to "fake" or "help" noise reduction by just reducing the volume when speech is not detected.
These are variants of squelch
- soft squelch where volume is reduced, but never totally silent (i.e. there is a setting for 
- fade out where the volume slowly tails off over say 30 secs when squelch closes instead of a hard cutoff, this can make hf squelch less jarring, and less likely to lose words during QSB.

f5bud, 2025-06-24:
I didn't know it, or even imagined it.
I did some research and understood the principle:
Divide the spectrum into two parts, one centered at 800Hz (a good portion of the voice), the other at 2000Hz (the noise).
When the two parts are roughly balanced, it means no one is talking.
I think this is a simplistic summary, but it's close.

sbridger, 2025-06-25:
The important feature is that there are syllabic pulse in the energy at 1-10Hz. That is the unique defining feature of SSB speech. 

To quote [Single Sideband Systems & Circuits pg292](https://archive.org/details/singlesidebandsy0000unse/page/292/mode/2up)

> A second property of speech is the low-frequency “syllabic” variations. These are in the range of 1 to 10 Hz or so. Figure 7.15 shows a schematic diagram that utilizes these variations. The signal is amplitude limited with diodes to reduce sensitivity to signal level, bandpass filtered at 600 Hz, envelope detected, and then bandpass filtered to recover the low-frequency syllabic variations that amplitude modulate the 600-Hz speech band. That is, the bursts of speech spectral energy within the 600-Hz bandpass filter vary at the syllabic rate. Noise and CW carriers are relatively weak in these syllabic frequency components.