f5bud/FX-4CR-V2 #54: Voice auto Call ends without indication

sbridger, 2025-08-16:
I just noticed that the voice call stops, but it gives me no indication that it has stopped, or that it is limited to 10 calls.
(The manual does tell me it is limited to 10 calls.)

There should be some indication of the state it is in.
e.g. 
change the string of the selected mode while calling to: "M.V: Calling every 15 seconds (3 of 10)"

priority: low.

sbridger, 2025-08-19:
OK tried it, but I find it rather un-intuitive.

The screen still shows 

```
SSB: Voice Beacon
M.V: Voice Beacon
A.B : Voice Beacon
CW: Voice Beacon
FM: Voice Beacon
```
But these all do the same thing.

Having the count and time in the status line is counter intuitive when you have the yellow autocall prompt in the main area. That's where I would expect to find the call N:delay be shown.

If you are going to have the several buttons trigger send, then they should force the default delay as they did before. I do find that sometimes I use 15sec pause and sometimes 1min pause alternating, and the button select is better than adjusting the knob every time.

e.g. it could be:

SSB : Send once
CW: knob selected (quick) pause, default 15 sec
AM/FM 30sec or perhaps a different knob selected (slow) pause, default 30sec.
A.B Not used for auto call, as I definitely need to A/B vfo switch repeatedly when calling CQ
M.V perhaps not used, as I sometimes use memories like A/B changing channels when calling CQ

RIT same as it is now. not used / cancel voice beacon / ptt  

The first 3 yellow lines could be:
```
SSB : Voice Beacon Once, (hold to exit)
CW: Voice Beacon <N> repeats <delay quick> seconds
AM.FM: Voice Beacon <N> repeats <delay slow> seconds
```

<N> and <delay> count down

As this will only be 3 lines on screen I suggest the last 2 lines are help:
```
Press and turn to adjust
AF: N repeats  TUNE: Delay (secs)
```

f5bud, 2025-08-20:
Okay, I was heavily inspired by your suggestions.

**"SSB: Send Once"**    Sends the message once
**"CW: Send x times every x seconds"**    Sends the programmed number of times and delays.

**(Press AF to change repaet and delay)**    Help message :)

A single press of AF displays the beacon settings menu.

I didn't do any more because it required a major transformation.
The code is common between SSB and CW beacons.

f5bud, 2025-08-20:
Ok, just changing mode stops the beacon

f5bud, 2025-08-21:
I just enabled beacon mode for AM/FM.
Same logic for all modes.
Will be available in the next update.

sbridger, 2025-08-31:
> I just enabled beacon mode for AM/FM.

Tested OK

closed 2025-08-31 by sbridger