f5bud/FX-4CR-V2 #59: CW TX Rising edge shape (key clicks)

sbridger, 2025-08-31:
I noticed on the scope that the rising edge of the RF in CW is a hard rise to 30% more than the output level, then an exponential fall to the correct level.
As opposed to [the expected 5ms linear ramp](https://www.ivarc.org.uk/uploads/1/2/3/8/12380834/keyclicks_version_1.pdf):

> The ARRL suggests that for a keying speed of
> 30wpm on a communications circuit with fading, a rise time of about 5ms is adequate 

Note that the sidetone to the speaker also suffers from audible keyclick in my opinion, and also should be shaped.

sbridger, 2025-09-05:
One way to do the sidetone, would be to double the dma sine table to make two complete sinewaves. Once the DMA pointer has passed from one sine wave to the next, you re-generate that sine wave (i.e. zero crossing to zero crossing). with the new amplitude (whilst the dma is working through the other table). i.e. it is buffered.

(again, the sidetone is not really important)

f5bud, 2025-09-29:
Come to think of it, I should be able to do something fairly easily.

The CW is generated by a 24kHz NCO, and I can use a sample block of 5.33ms to create a raised cosine ramp.
The same goes for the release.

```cpp
Complex SignalBase::ProcessTxCW(float32_t value)
{
	return mDdsMix.get() * ramp;		// ramp = raised cosine array
}
```

f5bud, 2025-10-01:
I added a CW signal shaper of the Raised Cosine type.
I also changed the internal organization of the filter switching, input signals, and delay management before the RF transmission begins.
The delays are now much shorter, around 3 x 5.33 ms ~ 16 ms.
The relays have a switching time of typically 3 ms, maximum 5 ms.

I didn't detect any noise or pulse during the switching.
This is a big improvement, as in the past there could be anomalies.

PS
Since the HF power can be adjusted during transmission, I thought it would be more logical to do the same for the power for "Antenna-tuner" (1 - 5 W)

[FX-4CR-V2-R-25-10-01.zip](https://github.com/user-attachments/files/22645187/FX-4CR-V2-R-25-10-01.zip)

Maybe a factor reset!